motioneye-project/motioneye #367: no local camera option

dsarf71, 2016-11-22:
Installing Motioneye on RPi3 Raspbian Jessie-lite with connected RPi camera.  Tested camera with 'raspistill'-- saw image.

Upon first login to <IP:8765>, local camera option was in dropdown.  I did not initially add bcm-2835-v4l2 to /etc/modules so I went back to do that.  After reboot, local camera option was gone.  Only Remote ME camera and Simple MJPEG options.  I have tried another reboot, motioneye upgrade, ensured camera enabled in raspi-config, and I have searched issues forum and have not found a fix.  

Output of dmesg | grep v4l below:
    [    3.965550] bcm2835-v4l2: scene mode selected 0, was 0
    [    3.971403] bcm2835-v4l2: V4L2 device registered as video0 - stills mode > 1280x720
    [    3.985796] bcm2835-v4l2: Broadcom 2835 MMAL video capture ver 0.0.2 loaded.

Thank you in advance!

ccrisan, 2016-11-22:
That happens when you don't have a (working) motion binary installed.

zglozman, 2017-06-14:
and whats the fix for that?

ccrisan, 2017-06-14:
Installing motion :)

ibravo, 2017-06-25:
I too tried the instructions from here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
on a clean Raspbian Raspberry Pi Zero W 

No luck.

ccrisan, 2017-06-26:
@ibravo what the command `motion -v` say?

ibravo, 2017-06-27:
This is the output:

`pi@garagepi:~ $ motion -v
motion: invalid option -- 'v'
motion Version 4.0.1, Copyright 2000-2016 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers

`

ibravo, 2017-06-27:
Solved. The camera was not connected correctly. Tested until I got a 1,1 with the following command:

$ vcgencmd get_camera
supported=1 detected=1

closed 2017-06-27 by ccrisan