motioneye-project/motioneye #369: RSTP protocol not supported

antikuza, 2016-11-25:
Hi. I can't add RSTP stream to Motion. I get error "protocol rtsp not supported"
Motion Eye Advanced Settings:
[image]

"motion -h" on server:
```
motion Version 4.0.1+git9df0c11, Copyright 2000-2016 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers
```

ccrisan, 2016-11-25:
Well your motion version in use is 3.2.12. You need to tell motionEye to use the 4.0 binary. Check your [motion_binary](https://github.com/ccrisan/motioneye/wiki/Configuration-File#motion_binary) setting.

teco13, 2017-10-10:
I have motioneye (not motioneye os) version 0.37.1 and motion 4.0.1 from https://github.com/Motion-Project/motion, but it does not work rstp (protocol rstp not supported)
[image]
[image]

Could you help me?
Thanks in advance

ccrisan, 2017-10-10:
It's rtsp, not rstp.

closed 2017-10-10 by ccrisan