motioneye-project/motioneye #370: Instructions for setting up a hub

wandering213, 2016-11-26:
I'm trying to connect a MotionEyeOS device to a server with MotionEye installed and I am getting authentication errors. Can you provide some instructions for the proper way to add a MotionEye camera to a hub?

ccrisan, 2016-11-26:
[This](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios) is the only doc on this matter. Make sure you run the latest motionEye(OS) versions on all units (even prereleases) in order to avoid compatibility issues.

wandering213, 2016-11-26:
I did see that and I have the latest version on both computers, what's the correct URL to use?

ccrisan, 2016-11-27:
The URL should be simply http://ip/ (motionEyeOS) or http://ip:8765/ (motionEye).

wandering213, 2016-11-28:
I give that a try, I get the following behavior. Authentication error and then the login prompt which won't accept any credentials until I refresh the page.:

[image]

[image]

My credentials are the same for both locations.

ccrisan, 2016-11-28:
Any special characters in that password? Have you tried changing it to something simpler?

wandering213, 2016-11-29:
Yes, that was it.I dropped the special characters and it worked.

closed 2016-11-29 by wandering213

ccrisan, 2016-11-29:
Please tell us which special characters you were using.

wandering213, 2016-11-29:
I used space and !.

ccrisan, 2016-12-04:
https://github.com/ccrisan/motioneye/commit/9d54c882477599c7fcf713998e2c2e9a5b92dccf fixes the use of spaces in passwords.