motioneye-project/motioneye #372: Apache https:\\mysite\sec  to http:\\localhost:8765

marcorighi, 2016-11-27:
sorry if the answer is already in this archive but I was not able to find it. I would like to send a request to my home apache (on dynamic DNS) https://myserver.adomain/sec and to get a rewrite to http://localhost:8965.

I try the following configuration

`<Location "/sec">
<IfModule !proxy_module>
	LoadModule proxy_module modules/mod_proxy.so
</IfModule>

<IfModule !proxy_http_module>
	LoadModule proxy_http_module modules/mod_proxy_http.so
</IfModule>

ProxyPass http://fileserver:8765
ProxyPassReverse http://localhost:8765

</Location>`

but I got only a redirection to motioneye page without password request and moreover I see some commands as a list (I forget the page format - css?- and the password). Moreover it is no visible noone connected cam.

Thank in advance for your help

chrisf4lc0n, 2016-12-25:
Dunno about the Apache, but here is a config for the lighttpd:

```
server.modules = (
  "mod_access",
  "mod_auth",
  "mod_proxy",
  "mod_simple_vhost",
  "mod_accesslog",
  "mod_cgi" )

fastcgi.debug = 1
server.document-root        = "/var/www"
server.upload-dirs          = ( "/var/cache/lighttpd/uploads" )
server.errorlog             = "/var/log/lighttpd/error.log"
server.pid-file             = "/var/run/lighttpd.pid"
server.username             = "www-data"
server.groupname            = "www-data"
server.port                 = 80

index-file.names            = ( "index.php", "index.html", "index.lighttpd.html" )
url.access-deny             = ( "~", ".inc" )
static-file.exclude-extensions = ( ".php", ".pl", ".fcgi" )

compress.cache-dir          = "/var/cache/lighttpd/compress/"
compress.filetype           = ( "application/javascript", "text/css", "text/html", "text/plain" )

include_shell "/usr/share/lighttpd/create-mime.assign.pl"
include_shell "/usr/share/lighttpd/include-conf-enabled.pl"

$SERVER["socket"] == "0.0.0.0:5556" {

     proxy.server  = ( "" =>
        (( "host" => "0.0.0.0", "port" => 8765 ))
    )

    ssl.engine = "enable"
    ssl.pemfile = "/etc/lighttpd/certs/lighttpd.pem"
}

```
You just have to create the *.pem file prior the setting it up

slokhorst, 2017-01-11:
You need to set [`base_path`](https://github.com/ccrisan/motioneye/wiki/Configuration-File#base_path).

chuma, 2017-03-07:
I have this setup working mostly OK now myself.  I set `base_path` to `/mypath` in `motioneye.conf` - NO trailing slash!

Then in an Apache config file:

> `<Location "/mypath">`
> `    ProxyPass http://localhost:8765`
> `    ProxyPassReverse http://localhost:8765`
> `</Location>`

Then I can load `http://server.name/mypath/` and motioneye comes up fine.  But the URL MUST have the trailing slash, otherwise the external assets (JS/CSS etc) do not load.

marcorighi, 2017-07-12:
Hi,
if I load http://server.name/mypath without trailing slash compares a strange page. Can you help me to force the trailing slash?

ccrisan, 2017-07-12:
Starting with 0.36, motionEye uses relative paths to access its resources. Therefore the `base_path` option is no longer needed (nor does it exist anymore).

In any case, this is not a valid motionEye issue but rather a matter of configuring the Apache webserver as a reverse proxy. Therefore I'll close the issue now.

closed 2017-07-12 by ccrisan