motioneye-project/motioneye #373: Run a command

mikkelson2, 2016-11-27:
Hello and sorry for my bad english.

I have installed motioneye over raspbian and have gotten it working fine with email alerts, but I don't seem to get python scripts working trought the "run a command" option. I have tested the scipts to be working and have tried to place them in /home/pi/ folder and run a chmod +x on them, but I never get them to work even thought I write in the blank space the right path to the script (/home/pi/sript.py). Is there any logfile where I could try to search the problem?

CatoAntonsen, 2016-11-29:
Wouldn't you write /path-to/python (space) (/path_ to/scriptfile)?

mikkelson2, 2016-12-01:
Thanks for the reply! I got it working by addind "python" before the script-path, so you were right! I was following a guide on one website and it had a picture where there was only the script-path added. So I thought that it would be enough and I had no prior experience from linux or python before this.

closed 2016-12-01 by mikkelson2