motioneye-project/motioneye #375: Linux Mint installation not working

championc, 2016-11-28:
Hi there,

I've followed the Ubuntu guide, taking note of the initial statement confirming it's non-test with Mint.  Anyway, I've proceeded and hopefully it's just something small which can be resolved quite easily.

Everything appeared to install perfectly without issue and I've gone back and re-run all install commands and each one has confirmed that all items are already installed.

However, when I run the command from the command line "meyectl startserver -c /etc/motioneye/motioneye.conf", I get the following

[code]
WebCam-Server bin # meyectl startserver -c /etc/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
    INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.35.1', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 256, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 436, in main
    run()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 394, in run
    application.listen(settings.PORT, settings.LISTEN)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1850, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 126, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 194, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
[/code]

ccrisan, 2016-11-28:
Well, given the error you receive:

    socket.error: [Errno 98] Address already in use

you probably have a motionEye server already running. Or some other service on that port.

championc, 2016-11-28:
But I cannot get anything if I try http://127.0.0.1

ccrisan, 2016-11-28:
How about http://127.0.0.1:8765?

championc, 2016-11-28:
That works !!!

Can I change the port number to 80 ?

ccrisan, 2016-11-28:
See https://github.com/ccrisan/motioneye/wiki/Configuration-File#port.

closed 2016-11-28 by ccrisan