motioneye-project/motioneye #376: USB Camera - Unable to open video device on a grey screen

championc, 2016-11-30:
Hi there
I am testing this software on a Linux Mint installation.  I get a grey camera screen with "Unable to open video device" written in the top left corner.

I have installed Cheese and this confirms that the USB webcam is working correctly.  I have tried another USB webcam and it has the same issue.  Both webcams are identified as V4L2 cameras.

chrisf4lc0n, 2016-12-08:
Have you installed motion?

championc, 2016-12-08:
Thanks for the reply.  However, I did install it.  I followed the Ubuntu installation process to a T.

Is there anything else that I can check ?  Or how can I check that Motion installed and is functioning correctly ?

chrisf4lc0n, 2016-12-16:
Try motion -h for the starters.

championc, 2016-12-18:
motion Version 3.2.12+git20140228

ccrisan, 2016-12-18:
Please upgrade to [motion 4](https://github.com/Motion-Project/motion/releases/tag/release-4.0.1)

championc, 2016-12-19:
:-(  How can I force upgrade from 3.1 to 4.0 please ?

WebCam-Server championc # pip install --upgrade motioneye
Requirement already up-to-date: motioneye in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pycurl in /usr/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: certifi in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: backports_abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: MarkupSafe in /usr/lib/python2.7/dist-packages (from jinja2->motioneye)
Requirement already up-to-date: six in /usr/lib/python2.7/dist-packages (from singledispatch->tornado>=3.1->motioneye)
WebCam-Server championc #

chrisf4lc0n, 2016-12-19:
You have to do it from the .deb package or build from source... Just Google
it.

championc, 2016-12-19:
Sorry, of course, I need to upgrade Motion, and not MotionEye !!!

championc, 2016-12-19:
Quick question - within the packages for Motion 4 for i386 are jesse, precise, trusty, xenial and jakkety.  Which do I use for my Linux Mint (Ubuntu) installation ?

chrisf4lc0n, 2016-12-19:
Which version of Mint?

championc, 2016-12-19:
v18 - Sarah

chrisf4lc0n, 2016-12-19:
Jessie.
Just do:
`cat /etc/debian_version`

championc, 2016-12-19:
championc@WebCam-Server ~ $ cat /etc/*-release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=18
DISTRIB_CODENAME=sarah
DISTRIB_DESCRIPTION="Linux Mint 18 Sarah"
NAME="Linux Mint"
VERSION="18 (Sarah)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 18"
VERSION_ID="18"
HOME_URL="http://www.linuxmint.com/"
SUPPORT_URL="http://forums.linuxmint.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/linuxmint/"
UBUNTU_CODENAME=xenial
cat: /etc/upstream-release: Is a directory
championc@WebCam-Server ~ $ cat /etc/debian_version
stretch/sid
championc@WebCam-Server ~ $

chrisf4lc0n, 2016-12-20:
Can you post the logs?

ccrisan, 2017-08-05:
I'm closing this due to lack of activity.

closed 2017-08-05 by ccrisan