motioneye-project/motioneye #377: Free Human Body recognition

MourIdri, 2016-12-03:
Guys, 

During hacakthon, I am using motioneye with watson for Human recognition. Since 250 images per day are free, It can be nice to share.. 
So this script is triggered when a motion is detected ( Motioneye functionality), then the image is sent to Watson analysis, and is uploaded to Bluemix cloud. Then It sends mail ( so I disabled Motioneye mail option) in the mail, there is the cloud url where the image is stored,  it also send a google hangout update for a given enduser. 
So you need a gmail account for sms and mail ( here https://accounts.google.com/SignUp?continue=https%3A%2F%2Fwww.google.fr%2F%3Fgws_rd%3Dssl%26pli%3D1&hl ) and a bluemix account for watson image recognition and easy 25gigs cloud storage ( here https://console.ng.bluemix.net/registration/ ).

Save it to a script.py file and it will 

`#!/usr/bin/python
# pip install boto3 boto watson-developer-cloud xmpp

import os
import json
import xmpp
import smtplib
import sys
import logging
import time
import datetime
import boto3
import boto.s3.connection
import boto.s3.connection
from os import environ
from watson_developer_cloud import VisualRecognitionV3
from os.path import join, dirname
from email.MIMEMultipart import MIMEMultipart
from email.MIMEText import MIMEText
from email.MIMEBase import MIMEBase
from email import encoders
from boto.s3.key import Key
from boto.s3.connection import S3Connection
from boto.s3.connection import OrdinaryCallingFormat
from boto.s3.connection import VHostCallingFormat
from boto.s3.connection import SubdomainCallingFormat

##############################INPUTS##############################
##path to image/capture files, were your cam is storing files
motioneye_source_path = '/var/lib/motioneye'
## Motion create a directory per day cwhen pictures are taken, so adding a change dire fucntion
now = datetime.datetime.now()
motioneye_pictures_subpath = now.strftime("%Y-%m-%d")
#newDirName = now.strftime("%Y_%m_%d-%H%M")
#print "Making directory " + newDirName
#os.mkdir(newDirName)
path = '%s/%s/' % (motioneye_source_path,motioneye_pictures_subpath)

#os.chdir(path)

##putting API after getting it from Bluemix portal at http://bluemix.com/ 250 analysis free per day
my_watson_apikey_forimage_recognition = '**********API_SERIAL_CAN_BE_GET_AFTER_BLUEMIX_REGISTRATION_ACCOUNT'
##putting connexion Data from  IBM cloud object storage S3 endpoint, can be ordered from http://bluemix.com/  25Gigas free
##and creating connection
credkeyid= '**********'
secretaccesskey= '**********'
hostendpoint = 's3-api.dal-us-geo.objectstorage.softlayer.net'
bucket_target = '**********'
##Connecting to object storage  :
s3conn = boto.connect_s3(aws_access_key_id= credkeyid,
                         aws_secret_access_key= secretaccesskey,
                         host = hostendpoint,
                         #debug=2,
                         is_secure=True,               # uncommmnt if you are not using ssl
                         calling_format = boto.s3.connection.OrdinaryCallingFormat(),
                         )
bucket = s3conn.get_bucket(bucket_target)

##charging the WATSON Image reconightion API
visual_recognition = VisualRecognitionV3('2016-05-20', api_key=my_watson_apikey_forimage_recognition)
filter_recognition = 'person'
##Details of account that will send alerts and account that will receive alerts XMPP alerts and mail
fromaddr = "themailfromchichthemailissent@gmail.com"
fromaddr_pswd = "**********passwordofthesendermail"
toaddr = "themailthatwillreceivethealerte@gmail.com"
smtpserver = 'smtp.gmail.com'
smtpserver_port = 587
##############################END INPUTS##############################

files = os.listdir(path)
for name in files:
        print(name)
        filepath= "%s%s" % (path,name)
##getting current directroy
        #cwd = os.getcwd()
        #print (cwd)
##removing extension of files to extract file details to play with json output
        base=os.path.basename(filepath)
        os.path.splitext(base)
        base2=os.path.splitext(base)[0]
        #print (base2)
        print (filepath)
##gathering path of each files
        with open(join(dirname(__file__), filepath), 'rb') as image_file:
##openning files and sending it to watson recognition and getting
                jsonfileanalyzed=json.dumps(visual_recognition.classify(images_file=image_file,threshold=0.1,classifier_ids=['classifier_1','default']),indent=2)
##uncomment the line below if you want to see the JSON file
                                #print(jsonfileanalyzed)
##saving json files for post-treatment and local operation based on conditions.
                out_file = open("filenametemp.json","w")
                json.dump(jsonfileanalyzed,out_file,indent=2)
                out_file.close()
                old_json_file = "filenametemp.json"
                new_json_file = "%s.json" % (base2)
                os.rename(old_json_file,new_json_file)
##parsing json file to search for key words.
                with open(new_json_file, 'r') as searchfile:
                        for line in searchfile:
                                if filter_recognition in line:
##if a person is reconized, a picture will be keept and an alerte will be sent
## I will be using the key word "person" from the json got from WATSON
                                        print ('There is a  person in this image')

## uploading data to IBM cloud object storage S3 endpoint
                                        key = name
                                        fn = "%s%s" % (path,name)
                                        # create a key to keep track of our file in the storage
                                        k = Key(bucket)
                                        k.key = key
                                        k.set_contents_from_filename(fn)
##we need to make it public so it can be accessed publicly
##using a URL like https://endpoint/bucket/image
                                        k.make_public()
                                        url="https://%s/%s/%s" % (hostendpoint,bucket_target,name)
                                        print('Uploading some data to {} with key: {}'.format(bucket_target, name))
                                        print('\nTry this URL in your browser to download the object:')
                                        print(url)

##Sending SMS or alerte suing an xmpp account or server:
                                        jid = xmpp.protocol.JID(fromaddr)
                                        cl=xmpp.Client(jid.getDomain(),debug=[])
                                        cl.connect()
                                        cl.auth(jid.getNode(),fromaddr_pswd)
                                        cl.send(xmpp.protocol.Message(toaddr,' ALARME !!!! movement of a person is detected by your motioneye!!!!', typ='chat'))
##Send mail
                                        msg = MIMEMultipart()
                                        msg['From'] = fromaddr
                                        msg['To'] = toaddr
                                        msg['Subject'] = "movement of a person is detected"
                                        body = "Please find the URL of the Image here : %s " % ( url)
                                        msg.attach(MIMEText(body, 'plain'))
##If you want to join image to this mail instead of having access to it using the web uncomment the line below
                                                                                #filename = name
                                        #attachment = open(filepath, "rb")
                                        #part = MIMEBase('application', 'octet-stream')
                                        #part.set_payload((attachment).read())
                                        #encoders.encode_base64(part)
                                        #part.add_header('Content-Disposition', "attachment; filename= %s" % filename)
                                        #msg.attach(part)
## details of target server
                                        server = smtplib.SMTP(smtpserver, smtpserver_port)
                                        server.starttls()
                                        server.login(fromaddr, fromaddr_pswd )
                                        text = msg.as_string()
                                        server.sendmail(fromaddr, toaddr, text)
                                        server.quit()
                                        os.remove(filepath)
                                        os.remove(new_json_file)
                                else :
                                        #delete picture and json file
                                        print ('no person in this image')
                                        os.remove(filepath)
                                        os.remove(new_json_file)

`

MourIdri, 2016-12-03:
[FACE_RECOGNITION_MAIL_XMPP_UPLOAD_IMAGE_TO_CLOUD.txt](https://github.com/ccrisan/motioneye/files/628936/FACE_RECOGNITION_MAIL_XMPP_UPLOAD_IMAGE_TO_CLOUD.txt)

closed 2016-12-14 by MourIdri

thundergreen, 2017-03-14:
@MourIdri I have a question: I wanna have a XMPP Upload with pictures / videos to my httpuplad ejabberd server or just a link to the file. Do you have an idea how to make it? Your script wont send it to an xmpp account

MourIdri, 2017-03-19:
This is what my script does but using hangout (gmail xmpp). To use jabber, You will have to change the script a little bit in order to change jabber account calls. ( else Use gmail, this more convenient).