motioneye-project/motioneye #378: Base Path for both Reverse Proxy and Direct Access?

MatthiasGemelli, 2016-12-04:
Hello CC & all,

I successfully setup reverse proxy following 
https://github.com/ccrisan/motioneye/wiki/Running-Behind-Nginx

Remaining Challenge: when I access MotionEye directly (on LAN, without proxy), the embedded scripts/etc don't load (same as through Reverse Proxy before changing base path).

Question: how can I configure MotionEye/NGINX to allow me to access the same camera both ways:
-via reverse proxy http://nginx-reverse.mydomain.com/cam4/   which proxy-forwards to:
- directly on LAN http://192.168.0.200/

And yes, maybe better to segment the cameras off and only go through the Reverse Proxy. 
In my case I go through 2: one hosted on AWS, one hosted on LAN. And each Proxy adds a /folder/ to the URL...

Thanks in advance for guidance...Matthias

PS: Seems I converted a dozen colleagues to MotionEyeOS...we'll do a hands-on session shortly. 
Great piece of software!! Respect, Crisan!

ccrisan, 2016-12-04:
@MatthiasGemelli I'm glad you find my work useful!

Unfortunately I don't think there's any way to make a motionEye instance available at both locations, as the URI (path) is an important element in the authentication mechanism.

The best thing you could do is to have it served at `/cam4` by both servers, which would need an nginx on the motionEye machine (instead of motionEyeOS you'll need to run Debian + motionEye).

Or you could have your existing nginx listen on the private interface (assuming you have a private network there) and serve the same thing, using the private IP of your nginx server :)

lebomb44, 2016-12-05:
Hi,
I did something like that in a previous project.
Instead of having obsolute path you only need relative path:
            <link rel="stylesheet" type="text/css" href="/cams/static/css/jquery.timepicker.css" />
should be changed to:
            <link rel="stylesheet" type="text/css" href="static/css/jquery.timepicker.css" />

ccrisan, 2017-05-28:
5ec72dc95b8acfdea25e3a8e9e939f952c34bc3c removes base_path setting. With @lebomb44's PR things should work well automatically behind a reverse proxy as well as when accessed directly.

closed 2017-05-28 by ccrisan