motioneye-project/motioneye #381: Email & webhook notifications have stopped

donnlee, 2016-12-07:
My motioneyeOS RPi2 was working great for 6 days. Then a few days ago, 12/5 23:16 US-PT localtime, motioneyeOS stopped sending email and stopped calling my webhook.

I see this in the motion.log:

```
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 280, in <module>
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 268, in main
  File "/usr/lib/python2.7/site-packages/motioneye/webhook.py", line 35, in main
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 34, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/httpclient.py", line 47, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 32, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/log.py", line 37, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/escape.py", line 38, in <module>
ImportError: No module named html.entities
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 280, in <module>
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 259, in main
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 33, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 46, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 32, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/log.py", line 37, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/escape.py", line 38, in <module>
ImportError: No module named html.entities
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 280, in <module>
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 268, in main
  File "/usr/lib/python2.7/site-packages/motioneye/webhook.py", line 35, in main
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 34, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/httpclient.py", line 47, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 32, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/log.py", line 37, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/escape.py", line 38, in <module>
ImportError: No module named html.entities
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 280, in <module>
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 259, in main
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 33, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/ioloop.py", line 46, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/concurrent.py", line 32, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/log.py", line 37, in <module>
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/escape.py", line 38, in <module>
ImportError: No module named html.entities
```

Other logs look ok; no scary messages.

If I do "Test email", mail is sent and received successfully.

ccrisan, 2016-12-07:
This is the first time I see this error. Might it be an SD card corruption problem? The respective module couldn't possibly get away from the filesystem, especially since we're talking about a read-only filesystem.

JamBalaya56562, 2026-11-07:
Closing, as this was on a 2016 motionEyeOS image and the only lead, as ccrisan suggested, was a corrupted SD card, since the failing module lives on a read-only filesystem there. If email or webhook notifications stop with a current motionEye, please open a new issue with the motionEye log from that time.

closed 2026-11-07 by JamBalaya56562