motioneye-project/motioneye #382: Stream Keeps freezing on Port 8081 with VLC and the Raspi home page doesn't have an image.

CBR900RRR, 2016-12-08:
Even while I was trying to download the log files, it would break connection and I'd have to time it for the connection to stay alive long enough for the file transfer to complete. not sure exactly when this started happening but it has been a couple weeks now for sure.

[messages (1).zip](https://github.com/ccrisan/motioneye/files/638504/messages.1.zip)

ccrisan, 2016-12-08:
Your messages.log file says:

    Dec  3 09:40:20 meye-07d70a40 user.notice brownout: low power supply voltage detected

and some of

    Dec  3 09:40:44 meye-07d70a40 user.notice wifi: disconnected for 20 seconds, rebooting

I recommend using a better power supply.

CBR900RRR, 2016-12-08:
thanks so much, sorry for wasting your time. Love the app. really makes it easy for us to use.

closed 2016-12-08 by ccrisan