motioneye-project/motioneye #383: device enable

RudySixOw, 2016-12-09:
Motioneye on debian VM version 0.35.1 motion version 4.0.1
Hi, sorry for my english, i have a problem when i change # @enabled on to # @enabled off and restart motioneye it don't take my changes but if i do that in web gui that works.
but when i do a diff between conf before and after there are no difference.
Do you have some idea?
thanks in advance

ccrisan, 2016-12-09:
1. Settings are cached in memory and therefore can only be changed via the user interface. If you must change that directly in the config file, do it while motionEye is *stopped*.
2. In addition to the `# @enabled` option, you'll also need to comment out the corresponding `include thread-x.conf` line in `motion.conf`.

closed 2016-12-09 by ccrisan

RudySixOw, 2016-12-09:
Ok thanks i don"t remove thread in motion.conf.