motioneye-project/motioneye #384: Duplicate cameras? Same video in different cameras

iSlide, 2016-12-11:
Good morning everyone, 
I'm a happy user of MotionEye, but I am encountering this annoying problem: 
I have 8 network cameras that work very well, some days I added the ninth without any problem except for the fact that his image I see also duplicated in number 7! 
I made many tests but when active the 9 video is duplicated on 7,  and if I turn off the 7 and relight the video of the 7 is duplicated in 9. 
I also check the Id in the various configuration files and I do not think there are mistakes. 
who can help me. 

thank you all
Have a nice Sunday

ccrisan, 2016-12-11:
Can you post your configuration files (motion.conf and thread-*.conf), after removing sensitive data?

iSlide, 2016-12-12:
Hi,
many thanks for your support!
Below you can find the configuration files.

motion.conf:
# @enabled on
# @show_advanced on
# @normal_password xxxx
# @admin_username admin
# @admin_password xxxx
# @normal_username user

thread thread-1.conf
thread thread-2.conf
thread thread-3.conf
thread thread-4.conf
thread thread-5.conf
thread thread-6.conf
thread thread-8.conf
thread thread-7.conf
thread thread-9.conf
webcontrol_html_output on
webcontrol_port xxxx
setup_mode off
webcontrol_localhost on

_______________
thread-7.conf:

# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 192.168.xxx.xxx
# @upload_username 
# @motion_detection on
# @upload_port 
# @upload_location 
# @preserve_movies 30
# @network_username xxxx
# @id 7
# @upload_password 
# @upload_method post
# @working_schedule_type during
# @webcam_server_resize off
# @network_password xxxx
# @upload_service gdrive
# @name xxxx
# @preserve_pictures 0
# @storage_device network-share
# @network_share_name xxxx
# @upload_enabled off
# @working_schedule

ffmpeg_output_movies on
stream_quality 85
threshold 15360
quality 85
noise_level 31
pre_capture 100
noise_tune on
stream_maxrate 5
output_pictures off
stream_localhost on
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes on
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
stream_port xxxx
rotate 0
stream_auth_method 0
lightswitch 0
framerate 5
emulate_motion off
snapshot_filename 
snapshot_interval 0
minimum_motion_frames 5
stream_motion off
target_dir /media/motioneye_192_168_xxx_xxx_xxxxx_xx/xxxx
text_double on
post_capture 100
stream_authentication user:xxxxx
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left xxxx
picture_filename 
locate_motion_style redbox
locate_motion_mode on
netcam_keepalive on
max_movie_time 0
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
ffmpeg_bps 768000
event_gap 5
netcam_tolerant_check on
netcam_url http://192.168.xxx.xxx:xxxx/videostream.cgi?user=xxxxx&pwd=xxxxx&resolution=32&rate=0

_____________
thread-9.conf:

# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @upload_username 
# @motion_detection off
# @upload_port 
# @upload_location 
# @preserve_movies 0
# @network_username 
# @id 9
# @upload_password 
# @upload_method post
# @working_schedule_type outside
# @webcam_server_resize off
# @network_password 
# @upload_service gdrive
# @name Camera9
# @preserve_pictures 0
# @storage_device custom-path
# @network_share_name 
# @upload_enabled off
# @working_schedule 

ffmpeg_output_movies off
netcam_userpass xxxxxx
stream_quality 85
threshold 1999
quality 85
noise_level 31
pre_capture 1
noise_tune on
stream_maxrate 5
output_pictures off
stream_localhost on
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness off
stream_port xxxx
rotate 0
stream_auth_method 0
lightswitch 0
framerate 2
emulate_motion off
snapshot_filename 
snapshot_interval 0
minimum_motion_frames 20
stream_motion off
target_dir /var/lib/motioneye
text_double on
post_capture 1
stream_authentication user:xxxx
on_picture_save /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save %t %f
on_movie_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end %t %f
text_left xxxx
picture_filename 
locate_motion_style redbox
locate_motion_mode off
netcam_keepalive on
max_movie_time 0
on_event_end /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start %t
ffmpeg_bps 43008
event_gap 30
netcam_tolerant_check on
netcam_url http://192.168.xxx.xxx:xxxx/videostream.cgi?user=xxxx&pwd=xxxx&resolution=32&rate=0

iSlide, 2016-12-12:
Hi,
many thanks for you support. 
In attachment I send to you the config files.

have  nice day
[config_files.txt](https://github.com/ccrisan/motioneye/files/645534/config_files.txt)

ccrisan, 2016-12-12:
The reason why I asked for your config files was to see if netcam_url points out to the same URL. You considered your local IP addresses as sensitive data and you removed them - making it impossible for me to check it.

iSlide, 2016-12-12:
I'm sorry.
The URL are:
netcam_url http://192.168.1.155:6055/videostream.cgi?user=xxxxx&pwd=xxxxx&resolution=32&rate=0

netcam_url http://192.168.1.158:6058xxxx/videostream.cgi?user=xxxxx&pwd=xxxxx&resolution=32&rate=0

A strange thing is that also the text on video is wrong "camera 7" or "camera 9" but is I click on video the names on the setup is correct.

Many thanks

iSlide, 2016-12-12:
Another think, if I copy the URL on browser I can see the correct video.

iSlide, 2016-12-16:
Hi Crisan,
I'm very sorry for the late of my reply, I solved the problem. I deleted both cameras and then I set the new cameras. 
Now it works fine!
Many thanks for your support!

closed 2016-12-16 by iSlide

kiat-huang, 2021-07-29:
I have the same problem.  With 6 cameras all different IPs after an hour or two one camera A started showing the images of another camera B. When I checked the config of A was unchanged.  But then the name also changed from, say, Camera A to Camera B.  The IP address in the motioneye webapp still showed as A, even though, with streaming it was clearly access the ip address of B.  I tried looking to see where motioneye keeps the camera configs, but had no joy.

  I'll look deeper - but if anyone knows, please say.  It will help editing the config files as text since I have common config options (Notifications, etc) that need to be identical and it is a hassle to go through the webapp for each camera.

starbasessd, 2021-07-29:
Usually it's a glitch in the camera-X.conf file. You'll find that Camera A has the same stream_port 808X as Camera B. the port # should be the same as the file camera-X.conf (808X)