motioneye-project/motioneye #385: lost picture - pink

lupinehorror, 2016-12-11:
hi all.
had this working fine but now the image has become very distorted...and pink! (please see attached).
anyone have any ideas? rebooted, changed the resolution, altered the frame rate...still nothing.
[image]

fireheadman, 2017-01-14:
I've had issues similar to this....  Last time it occurred it was due to a USB camera going out.   I replaced the camera.     Another time I found out my Powered USB Hub was going out and not providing enough power for 2x camera + the PI.   I moved the PI to its own power source and this cleaned up the picture.    And lastly... you might want to baby step the resolution, starting with the smallest resolution and moving up one at a time.  I use this chart to help me   https://en.wikipedia.org/wiki/Display_resolution#/media/File:Vector_Video_Standards8.svg

JamBalaya56562, 2026-11-09:
fireheadman's reply points to the usual causes of this picture: a failing USB camera, or a hub that does not give enough power. There was no follow-up since 2017, so closing. Please open a new issue if this happens on a current version with a known-good camera and power supply.

closed 2026-11-09 by JamBalaya56562