motioneye-project/motioneye #386: DropBox

rfv-370, 2016-12-11:
Hi Crisan,

My config-cam:  root-cam directory is a samba share with upload media/picture/movie to dropbox.

1) permanence of files: I understand even if we set upload for pictures, movies, etc... these will stay anyway on the location of origine ie- root directory, after being uploaded?

2) Root-cam directory behavior: What happens if the samba share is lost for the root directory? Does motioneye revert to /var/lib/motioneye temporarily and then back to the samba share?

3)Dropbox-link: Each time my router reset - which happens more than once per day, the dropbox link crash and most of the time I need to reboot the Pi as well - and sometimes reset the dropbox link. Any idea of a workaround?

4) Re- Movie lifetime: inactive?  I understand that motioneye will not delete any files neither on camera-root directory or dropbox - so we would have a separate Cronjob to do that?

best

Robert

ccrisan, 2016-12-11:
1. I don't understand the question. motionEye will not move your media files once they have been uploaded.
2. If any storage error occurs, the motion daemon will complain and will keep trying to save the media files in the configured location.
3. What do you mean your "dropbox link crashes"? The dropbox "link" in motionEye is based on a RESTful API that doesn't care if your internet connection has been re... booted.
4. motionEye will *never* delete files from your Dropbox, regardless of that setting. This is a security measure.

rfv-370, 2016-12-11:
Thanks.
Regarding point 3 ; well it seems unstable, sometimes it uploads sometimes not so I am trying to understand why... - does the uploadservices.json can be empty...? I tried to empty it and re-force the dropbox link. but the .json has not been updated... albeit it was uploading...

Where is the motioneye.log ?
I looked in /var/log and /var/log/motion but did not manage to find the log.

ccrisan, 2016-12-12:
uploadservices.json should not be empty. If you can reproduce the bug starting with a fresh install, please  list all the steps here, please.