motioneye-project/motioneye #387: Frame stuck/motion detected when creating a new video

xmflsct, 2016-12-12:
I am running motionEye on my RPi3 with normal Raspbian. I turned off text overlay, video streaming to reduce load. I am taking in a webcam stream of 720p with reduced 10fps, and I enabled continuous recording with motion detection functions.

The problem is that, I get quite some motion detection false alrams, and I find out that the alarm time is exactly the same when motionEye starts recording of a new video file (h.264 mp4). I set a limit of 15 minutes just so that it won't be a super large video file.

Why is this happening? Is it because RPi3 is not powerful enough?

ccrisan, 2016-12-12:
It might be that the motion daemon starts a new event as soon as the max movie time limit is reached (and therefore a new movie is created).

I'm sorry but there's nothing motionEye can do about this. I guess you could either use motionEye as a security camera with alerts or for continuous recording, but not both.

closed 2016-12-12 by ccrisan

xmflsct, 2016-12-13:
Thanks @ccrisan for the reply! That might be the case indeed. Hope there will be a workaround soon. :)