motioneye-project/motioneye #388: Hot Button

moonguar, 2016-12-12:
Calin, This is more a request than a problem.  Is there anyway to build a button that would turn motion off on all cameras at once?

thanks,  Kent

chrisf4lc0n, 2017-01-15:
[https://github.com/ccrisan/motioneye/wiki/Action-Buttons](https://github.com/ccrisan/motioneye/wiki/Action-Buttons)

moonguar, 2017-01-15:
Calin, I was referring to a hot button on the motioneyeOS side to turn off motion on all cameras at once. Based on the wiki link you provided, it won't help because you would still have to disable motion on each individual camera which doesn't make disabling all cameras at once any easier.

ccrisan, 2017-01-16:
@moonguar there's no such functionality yet. Sorry about that.

JamBalaya56562, 2026-11-09:
This asks for the same as #192: one control that switches motion detection on or off for all cameras at once. Today it is still per camera only. Closing this one as a duplicate, so the request is tracked in #192.

closed 2026-11-09 by JamBalaya56562