motioneye-project/motioneye #390: Unable to open video device

sapcons, 2016-12-13:
All of a sudden my cameras turned grey with the above error message. When I select the camera I can see the fps values are changing. Network connection is stable. Is there any file I can search for the root cause?

xmflsct, 2016-12-13:
I assume ffmpeg is the core used for streaming videos.

I had a similar issue, that motionEye says cannot connect to camera. Then I use ffmpeg try to connect to the stream, and then I can debug from there. Correct me if I am wrong @ccrisan.

sapcons, 2016-12-13:
Thanks for your help. The point is that I didn't change a thing. I am currently upgrading pi and motioneye to see whether this can fix the problem...

xmflsct, 2016-12-13:
No worries. :) But run ffmpeg, you will be able to see more details, at least what is the error message of connecting.

sapcons, 2016-12-13:
Can you please explain a bit more in detail? Thanks

xmflsct, 2016-12-13:
Example:
`ffmpeg -loglevel debug -i "rtsp://yourcamera" /saveto/localfile.mp4`
With loglevel to **debug** or even to **trace**, I could see what is the error message.
Reference: [https://trac.ffmpeg.org/wiki/StreamingGuide](FFmpeg Streaming Guide)

sapcons, 2016-12-14:
Couldn't find the root cause. Restored my backup and everything is running ok now. Thanks for your help.

closed 2016-12-14 by sapcons