motioneye-project/motioneye #391: Attached item in mail notification

Odwyn, 2016-12-16:
Hello,

Please, can you add the possibility to have an attached item on notification email.
Like the last picture during motion.

Thanks

ccrisan, 2016-12-16:
This is already possible. See https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails.

closed 2016-12-16 by ccrisan

FredClement91, 2019-08-21:
Hello,
It doesn't work for me. I see that he calls a url looking like "/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" picture_save 1 /home/video/picture.jpg"

I first found out that maybe some curl version don't support the space in "Content-Type:application/json", but that was a simple fix. Anyway, the system is still not sending pictures.