motioneye-project/motioneye #392: Storage Device

Odwyn, 2016-12-16:
Hello,

i can't open a request.
Can you add a storage device like SSH path, or a webdav where storage picture or movies ?

Thanks.
Regards

chrisf4lc0n, 2016-12-17:
No need for the SSH... You can use the SSHFS to map your remote drive locally and then just point MotionEye to save files to it.

ccrisan, 2016-12-17:
Upload via SSH (SFTP) is planned. Webdav, nope (I don't have experience with that). Until then, as @chrisf4lc0n said, sshfs is your option.

hellyberry, 2017-01-27:
rsync via ssh is also currently possible and very powerful, and low on system-resources, to automatically delete files once transferred and verified,..

ericjanvanputten, 2017-03-03:
Is storage like network drive as in motioneyeos possible as well?
For more computing power i divided across multiple devices, to now encounter that non-os doesnt have network disk available... 

still cool solution :)

ccrisan, 2017-03-04:
@ericjanvanputten set [smb_shares](https://github.com/ccrisan/motioneye/wiki/Configuration-File#smb_shares) to `true` and make sure your motionEye runs as root.

ericjanvanputten, 2017-03-04:
@ccrisan and i was adding packages and looking into hard mounting... silly me ;) 
thanks man!

closed 2019-06-21 by ccrisan