motioneye-project/motioneye #393: Is there a way a change motion log level?

xmflsct, 2016-12-16:
I have a webcam that has buggy implementation of rtsp protocol. I use VLC as a proxy, trying to solve most of the issue. Still there will be some time that when VLC cannot solve data issue, then the motion is barking crazy, creating enormous log file in /var/log/ on motion.log, syslog and user.log. I understand it might be a motion issue, but I see that when I start motion, I can decrease its loglevel. Is there way to set this in motioneye? Thanks!

xmflsct, 2016-12-16:
I figured it out. In `/etc/motion/motion.conf` I can control the log level and log type. For now, it won't bark anymore. 👍

ccrisan, 2016-12-16:
That file is ignored by motion when ran using motionEye. You can't tweak the motion log level with motionEye, unfortunately. What motion version do you use?

xmflsct, 2016-12-16:
I have 4.0.1 installed. You are right, it doesn't actually work.

Can you maybe point me to the script that calls motion? Maybe I can pass a parameter there when running with motionEye. Is that an idea?

ccrisan, 2016-12-16:
See https://github.com/ccrisan/motioneye/blob/master/motioneye/motionctl.py#L108. You could set your motionEye log level to `warn` or `error` which will in turn decrease the motion log level.

xmflsct, 2016-12-16:
Perfect! I will try it now.

xmflsct, 2016-12-16:
By the way, I try to modify `/ect/motioneye/motion.conf` as well for log_file to `/dev/null`, as I can see that this is the config file that is used. It doesn't seem to work. I will continue monitoring it for a while and see what happens. Thanks for your help so far!

ccrisan, 2016-12-16:
That is the file being used, but the log level is being supplied via command line and therefore the file setting is ignored.

closed 2016-12-16 by ccrisan