motioneye-project/motioneye #394: CPU 100% on mint 18 (Ubuntu 16.04)

mkauzlar, 2016-12-17:
Hello, there seems to be a problem with the recent update of ffmpeg on Mint 18.
after the update the process:
/usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
goes to 100% CPU and everything becomes deadly slow (unusable).
The thing about ffmpeg is just a guess as the update installed also  a bunch  of other software and dependencies as well.
Anyhow the update is causing it.
I think before ffmpeg was 2.8.6-1ubuntu2 and after 2.8.10-0ubuntu0.16.04.1

chrisf4lc0n, 2016-12-17:
Can you downgrade?
[http://www.howtogeek.com/117929/how-to-downgrade-packages-on-ubuntu/](http://www.howtogeek.com/117929/how-to-downgrade-packages-on-ubuntu/)

mkauzlar, 2016-12-17:
I can't because it's messy, but, it's a new box so I will reinstall Mint and will lock some packages from upgrade. If that doesn't work I'll try another distro.

chrisf4lc0n, 2016-12-17:
I currently stay away from 16.04, even though there have been many improvements many things have been messed up, so until I will get forced to move I will stick to 14.04.

mkauzlar, 2016-12-17:
I will try with 14.04 and report back

chrisf4lc0n, 2016-12-17:
Hold on, I have just rediscovered that I have built from source:
`chrisf4lc0n@Ubunciak:~$ ffmpeg -version
ffmpeg version N-76840-g5f2c233 Copyright (c) 2000-2015 the FFmpeg developers
built with gcc 4.8 (Ubuntu 4.8.4-2ubuntu1~14.04)`

chrisf4lc0n, 2016-12-17:
I still have got a source on my server... I will try to find out where did I get it from...

mkauzlar, 2016-12-19:
the problem of 100% cpu went away after installing ffmpeg 3.2 using this ppa:

sudo add-apt-repository ppa:jonathonf/ffmpeg-3

now the load is spread across all cores (4) which run at about 30-50% eeach. The performance of the video is still low though (max 9fps) and sluggish.
For real time monitoring is it possible to skip ffmpeg and use some sort for pass through streaming?
For instance if I use guvcview to stream the 3 cameras at the same time I get about 40fps on each.
We need the monitor to monitor real time movements of 3 cameras (15fps will suffice).

chrisf4lc0n, 2016-12-19:
What CPU have you got in there, if you don't mind me asking?
With the ffmpeg which I have compiled I do not seem to be getting more than
10% CPU usage with all the other services running in the background.
I do however have installed a semi-decent graphic card, but I am not sure
if ffmpeg uses it...

mkauzlar, 2016-12-19:
the CPU is low end. I figured out that the problem is in the rendering.
If I connect from a different machine (my workstation with high end cpu and gpu) the fps is ok.
If I connect locally the web browser struggles to render webcams windows from the application.
I installed zoneminder now and I can get 30fps on the local web browser which is 3 times the performance get from motioneye.
I'm not an expert in motion though, and I guess things can be tweaked perhaps to get better performance.

chrisf4lc0n, 2016-12-19:
Having the picture in browser renders a different problems. It would not be just the motion or motioneye having problems with it... Browser you are using can also contribute to the problem, on top of that you are trying to run a desktop environment on the machine which is running dozens of processes in the background.
I would personally stick to headless for motion/motioneye and view it from another machine...
I hope this makes sense.

mkauzlar, 2016-12-20:
The browser is firefox and I agree that it would  be better to use a different machine for rendering however this deployment requires the machine to display directly as it's located in a surgery room.
The machine is fully capable of displaying 4k videos, 3 webcams or anything. It actually works displaying them directly on the web browser when using ZM. I tried with different browsers and the result is the same.
But this issue was reported about the 100% CPU core usage which I suppose was due to ffmpeg version.
So this issue can be closed now.

chrisf4lc0n, 2016-12-20:
I know it might be a long shot and might be rather messy, but I would give it a go with different ffmpegs...

closed 2017-08-05 by ccrisan