motioneye-project/motioneye #395: Temporary solution to triggering recording

chrisf4lc0n, 2016-12-17:
Hi Calin,
I did not want to start a new thread about it as there have been many questions regarding this, but I have to ask you if thread-*.conf gets overwritten each time motioneye/motion start? If they do, is there a simple way of stopping that?

The reason for that question is, that I may have a rather awkward work around to give people a button which will trigger recording per camera.

It would be possible to create a bash script which replaces the emulate_motion off to emulate_motion on
`sed -i '/emulate_motion off/c\emulate_motion on' /etc/motioneye/thread-1.conf`
Restarts the motioneye to activate recording and another bash script which would do exactly the opposite.

Both scripts could be activated by the "IFTTT Do Button" from any Andoird/IOS device.

ccrisan, 2016-12-18:
Unfortunately there's no way to prevent motionEye from overriding that setting as it is used by the *continuous recording* and the *all frames* still images setting.

But you could use the motion http control interface to alter this setting in real time (the control port when running under motionEye is 7999).

chrisf4lc0n, 2016-12-18:
Thanks, I will look into that.

chrisf4lc0n, 2016-12-18:
OK, I think I have got it.
I could use curl to post required action locally through the http direct motion request, like _snapshot_ or _makemovie_

chrisf4lc0n, 2016-12-18:
Sending any instructions through the command line like:
`/usr/bin/lwp-request http://localhost:7999/2/action/snapshot
`
Does work but crashes the camera in motioneye ;(
I am closing this thread until I may come up with anything else.
As always thank you very much for all your help @ccrisan

closed 2016-12-18 by chrisf4lc0n

ccrisan, 2016-12-18:
I remember trying this snapshot function myself and I also remember it crashing the motion daemon. This needs to be investigated upstream (in motion). I hoped this would get fixed with 4.0 but it seems it's not the case.

nachonacho2, 2016-12-29:
Hi all,
I have a somewhat related query and a request for help.
I am trying to save a snapshot using an action button, however I get the error message "An error occurred. Refreshing is recommended." when I press the action button. The script works when executed from the terminal command line but produces only the error message from the motioneye web interface. Any suggestions?
Code (note, URL = the snapshot url provided by motioneye):
`
#!/bin/bash

DATE=$(date +"%Y-%m-%d_%H%M%S")
curl 'URL' -o "/home/pi/webcam/$DATE.jpg" -create-dirs
`

Cheers,
Nacho

ccrisan, 2016-12-29:
Please attach your log files.

chrisf4lc0n, 2016-12-29:
This is really strange, because I cannot see anything in your script which might be potentially affecting the functionality of MotionEye.

nachonacho2, 2016-12-30:
Thanks for the quick replies.
ccrisan, the log is below. This covers the period of logging into the web interface as admin, turning on the camera in the sidebar, pressing the action button twice (at 19:31:04 and 19:31:10), and then turning off the camera in the sidebar and logging out.

chrisf4lc0n, MotionEye otherwise continues to function, just the action button does not exhibit the intended behaviour of saving the snapshot. The script works when executed from the command line and motion continues to run with no errors.

Log:

> Dec 30 19:30:13 raspberrypi meyectl[394]: WARNING: 403 GET /login/?_=1483086613058&_username=_&_signature=25a4e23dfbde1c1eecdea98aac902cd89b73df63 (192.168.1.100) 3.36ms
> Dec 30 19:30:13 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Fri Dec 30 19:31:13 2016 [try http://www.rsyslog.com/e/2007 ]
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [WRN] [ALL] thread config option deprecated use camera
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
> Dec 30 19:30:24 raspberrypi motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
> Dec 30 19:30:24 raspberrypi motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
> Dec 30 19:31:04 raspberrypi meyectl[394]: ERROR: [Errno 2] No such file or directory
> Dec 30 19:31:04 raspberrypi meyectl[394]: Traceback (most recent call last):
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1425, in _stack_context_handle_exception
> Dec 30 19:31:04 raspberrypi meyectl[394]: raise_exc_info((type, value, traceback))
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1627, in wrapper
> Dec 30 19:31:04 raspberrypi meyectl[394]: result = method(self, *args, **kwargs)
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1598, in post
> Dec 30 19:31:04 raspberrypi meyectl[394]: self.run_command_bg(command)
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1601, in run_command_bg
> Dec 30 19:31:04 raspberrypi meyectl[394]: self.p = subprocess.Popen(command, stderr=subprocess.STDOUT, stdout=subprocess.PIPE)
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
> Dec 30 19:31:04 raspberrypi meyectl[394]: errread, errwrite)
> Dec 30 19:31:04 raspberrypi meyectl[394]: File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
> Dec 30 19:31:04 raspberrypi meyectl[394]: raise child_exception
> Dec 30 19:31:04 raspberrypi meyectl[394]: OSError: [Errno 2] No such file or directory
> Dec 30 19:31:04 raspberrypi meyectl[394]: ERROR: 500 POST /action/1/preset1/?_=1483086663642&_username=admin&_signature=66cddeed933e92aa97c530bfa1003dd01f1b8643 (192.168.1.100) 27.90ms
> Dec 30 19:31:10 raspberrypi meyectl[394]: ERROR: [Errno 2] No such file or directory
> Dec 30 19:31:10 raspberrypi meyectl[394]: Traceback (most recent call last):
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1425, in _stack_context_handle_exception
> Dec 30 19:31:10 raspberrypi meyectl[394]: raise_exc_info((type, value, traceback))
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1627, in wrapper
> Dec 30 19:31:10 raspberrypi meyectl[394]: result = method(self, *args, **kwargs)
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1598, in post
> Dec 30 19:31:10 raspberrypi meyectl[394]: self.run_command_bg(command)
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 1601, in run_command_bg
> Dec 30 19:31:10 raspberrypi meyectl[394]: self.p = subprocess.Popen(command, stderr=subprocess.STDOUT, stdout=subprocess.PIPE)
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
> Dec 30 19:31:10 raspberrypi meyectl[394]: errread, errwrite)
> Dec 30 19:31:10 raspberrypi meyectl[394]: File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
> Dec 30 19:31:10 raspberrypi meyectl[394]: raise child_exception
> Dec 30 19:31:10 raspberrypi meyectl[394]: OSError: [Errno 2] No such file or directory
> Dec 30 19:31:10 raspberrypi meyectl[394]: ERROR: 500 POST /action/1/preset1/?_=1483086670015&_username=admin&_signature=7f208b554e114264c9bcdaae8524ca3b1bd81be2 (192.168.1.100) 26.36ms
> Dec 30 19:32:20 raspberrypi meyectl[394]: WARNING: 403 GET /login/?_=1483086739647&_username=_&_signature=e466d6ace268524a61b17eef2458f4ddb1aefffc (192.168.1.100) 3.19ms
> Dec 30 19:32:20 raspberrypi rsyslogd-2007: action 'action 17' suspended, next retry is Fri Dec 30 19:33:20 2016 [try http://www.rsyslog.com/e/2007 ]
> Dec 30 19:32:28 raspberrypi vncserver-x11[541]: Connections: connected: 192.168.1.100::60005
> Dec 30 19:32:28 raspberrypi vncserver-x11[541]: Connections: authenticated: 192.168.1.100::60005, as (anonymous) (d permissions)
>

ccrisan, 2016-12-30:
1. Next time, please *attach* the log file rather than pasting it here and making it very difficult for us to read.

2. The root cause seems to be that the file is actually missing:

        raspberrypi meyectl[394]: OSError: [Errno 2] No such file or directory

    What action type have you chosen? What is the full path to the script associated to the action?

nachonacho2, 2016-12-30:
Sorry about that!

Strange that it thinks the file is missing given the action button only appears because the file is present.
I used preset1. The full path is /etc/motioneye/preset1_1

nachonacho2, 2016-12-30:
Debug log file attached.
[motion.txt](https://github.com/ccrisan/motioneye/files/678618/motion.txt)

chrisf4lc0n, 2016-12-30:
What sort of button are talking here about? Have you tested that the button itself actually works? Any chance I could see a full script monitoring the button press?

nachonacho2, 2016-12-31:
Action buttons. See here:
https://github.com/ccrisan/motioneye/wiki/Action-Buttons
https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

ccrisan, 2016-12-31:
This is motion.log. i need motionEye.log.

nachonacho2, 2016-12-31:
Sorry for the delay; motionEye.log is not being created (that I can see).
To confirm, I changed the log directory path in motioneye.conf to /home/pi/webcam but only motion.log is  then written to this folder.
I have uninstalled and reinstalled motioneye with no change.
For clarity, the original log data that I pasted in my second post was from syslog.

Not sure where to go from here... happy to try any suggestions :)

ccrisan, 2016-12-31:
Go to syslog. If you're on systemd, motioneye log can be obained as follows:

    journalctl -u motioneye > motioneye.log

nachonacho2, 2017-01-01:
Calin, thank you very much for your help. I have managed to resolve the issue. For completeness sake, the motioneye log and my script are attached.
The problem was I had edited the script file in windows and there was a carriage return causing the "No such file or directory" error. As per this advice ( http://unix.stackexchange.com/questions/189254/no-such-file-or-directory-but-i-can-see-it ) I used dos2unix to fix the preset file and the action button now works.
Cheers!

[preset1_1.txt](https://github.com/ccrisan/motioneye/files/679901/preset1_1.txt)
[motioneye.txt](https://github.com/ccrisan/motioneye/files/679902/motioneye.txt)