motioneye-project/motioneye #396: Record error

ouafnico, 2016-12-20:
Hello

When trying to record H264 continuously

Dec 20 08:59:58 pcduino3nano motion[4828]: [1] [ERR] [ENC] get_oformat: ffmpeg_video_codec option value mp4 is not supported
Dec 20 08:59:58 pcduino3nano motion[4828]: [1] [ERR] [EVT] event_ffmpeg_newfile: ffopen_open error creating (new) file [/var/lib/motioneye/Camera1/video-08-59-57]:
Dec 20 08:59:58 pcduino3nano meyectl[2774]: ERROR: connection problem detected for mjpg client for camera 1 on port 8081
Dec 20 08:59:58 pcduino3nano motion[4828]: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Dec 20 08:59:58 pcduino3nano meyectl[2774]: WARNING: Connect error on fd 20: ECONNREFUSED

The folder Camera1 exists, and not right problem.
OS is ARMBIAN (Debian8).

ccrisan, 2016-12-20:
What motion version is this?

ouafnico, 2016-12-20:
I'm not attached to a specified distribution for this board. It works on Arch so I moved out Armbian (but it was lasted versions provided by armbian repo).

closed 2016-12-20 by ouafnico