motioneye-project/motioneye #397: GPU motioneye?

ouafnico, 2016-12-20:
Hi

Sorry it probably has been asked.
Is there a MMAL working version, to use GPU for boards like Pi?
I have compiled ffmpeg with mmal, but not better, still movies with low FPS.
When installing motion-mmal the recording does not work.

(I'm using arch).

Any "official" solution?

ccrisan, 2016-12-21:
There's some work on this direction (motion 4 has support for the Raspberry PI MMAL interface) but I don't think it uses it to encode movies; I might be wrong though.

ouafnico, 2016-12-21:
What are you using on motioneye to encode?

ccrisan, 2016-12-21:
motionEye doesn't encode anything. motion does all the encoding stuff and as far as I know it uses the ffmpeg libs for that.

ouafnico, 2016-12-21:
Ok.. i tried mmal motion but got errors on encoding. When I use motion 4 it does not use gpu. Too bad :-)
Thanks for the info

closed 2016-12-21 by ouafnico