motioneye-project/motioneye #398: bcm2835-v4l2 error 0 waiting for frame completion

ecq123, 2016-12-20:
I found an error in the Kern.log, bcm2835-v4l2 error 0 waiting for frame completion.
I have a fresh-built Raspbian with the latest Kernel. Motioneye works fine and so does everything else. The bcm2835-v4l2 error occurs randomly, which the camera will suddenly stop working for 4 sec and works again. I have the continuous recording option on, with movie length of 3600 seconds. Usually, there are 24 media files per day, however, with such error, sometimes it will get up to 40/50 files. There's no problem if I am using MotioneyeOS, but with Motioneye daemon, it have the error. I am wondering if anyone has encountered such problem?

I am using Raspberry Pi 3 Model B

Thanks

chrisf4lc0n, 2016-12-21:
Whenever updating to the newest kernel there is a chance of incompatibility, have you tried to revert to the previous kernel version?
What kernel version are you using and on what Pi?
I have just gone through the kern.log on my Pi 3 with 4.4.38-v7+ #938 Kernel and I cannot see any of that.

ecq123, 2016-12-21:
That's the Kernel version I have, 4.4.38-v7+ #938
The problem occurs randomly. I left my Pi 3 running for 24 hours, it often had 10 - 15 errors. Each time the error will cause the recording to stop, and restart in a couple of seconds.
I can try to downgrade the Kernel to the oldest that the Pi 3 can support, and see if the error still exists.

Thanks

chrisf4lc0n, 2016-12-21:
Very often hardware failures can contribute to stuff like that. What are
yours CPU temps and what PSU are you using.

ecq123, 2016-12-21:
I am currently using Kernel version 4.4.9-v7+ #884 and still encounters the problem.

My CPU temperature is around 55C and the power supply is a 5V, 1A AC to DC micro USB wall plug.
I am not sure if those are the things that cause the problem. When I use MotioneyeOS (with the same specs), it has no problem. Then I switched to Motioneye, with Raspbian OS installed on top, then the problem occurred.

Thanks

chrisf4lc0n, 2016-12-21:
Mind that MotionEyeOS runs fewer processes in the background, for the RPI 3
it is advisable to have 2.5A.

ecq123, 2016-12-21:
Right, that's why I wanted to see which process is conflicting with Motioneye.

chrisf4lc0n, 2016-12-21:
They are not in conflict, they might be just using too much power...

ecq123, 2016-12-21:
I think I have found the cause of the problem. I disabled the NTP service, everything works fine. It has been running for 3 hours without any error. I'll keep testing it and let it run for at least 24 hours, and see if the error has complete gone or not.

ecq123, 2016-12-22:
So, there's another error I found in motion.log

[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
[0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful,killing it!
[0:motion] [ERR] [ALL] main: cleaning Thread 1
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

closed 2017-01-02 by ecq123