motioneye-project/motioneye #399: Streaming resolution

ivalion2133123, 2016-12-21:
Hello is it possible to stream video from camera in lower resolution (Streaming Resolution) but without resolution reduction in my own gui?

chrisf4lc0n, 2016-12-21:
It is it will get upscaled, but the quality will not be that great.

ivalion2133123, 2016-12-21:
Can't understand you what i want is to stream with low resolution to other motioneye instance but the resolution in the original motioneye must stay the same.

At the moment it also chenges.

chrisf4lc0n, 2016-12-21:
Oh, sorry now I get you. I think ;)
You just want to change the Streaming Resolution under Streaming Tab?
You can reduce the quality, but no resolution at the moment.

chrisf4lc0n, 2016-12-21:
Are you using MotionEye or MotionEyeOS?
If you are using MotionEye installed on another system, you can make the lighttpd or any other web server to catch the stream, reduce the resolution and stream it back out on a another port.

ccrisan, 2016-12-21:
@ivalion2133123 you can reduce the streaming resolution using Preferences/Resolution Dimmer. It will scale down the streaming resolution proportionally to what you choose there, leaving the capture resolution untouched.

closed 2019-06-21 by ccrisan