motioneye-project/motioneye #400: Allwinner H3 devices support

ivalion2133123, 2016-12-21:
Hello motioneye works well on H3 devices (orange pi, nano pi etc).

What i would like to use is a [special version of ffmpeg that does hard-encoding on H3](https://github.com/avafinger/ffmpeg_cedrus264_H3).

The command is then sudo ./ffmpeg -f v4l2 -channel 0 -video_size 640x480 -i /dev/video0 -pix_fmt nv12 -r 30 -b:v 64k -c:v cedrus264 test.mp4

Is it possible to add support? Where are the ffmpeg encoding options located?

ccrisan, 2016-12-21:
I don't know what you want me to answer here. The ffmpeg libs (not the binary!) are used internally by motion to encode movies. They are usually at /usr/lib. If you have compiled a custom ffmpeg version, make sure you have configured it with `--prefix=/usr` and that you have installed it. From your message I believe you're running the ffmpeg binary directly from your home directory, with sudo, which is not of any use for motion.

ivalion2133123, 2016-12-26:
The command above was only an example to show that it is possible to encode video from camera using video acceleration.

Now it's a question if i can do it also in motioneye to free cpu cores.

ccrisan, 2016-12-26:
If ffmpeg libraries are accelerated, then motion will be accelerated as well :)

Nzt2, 2017-05-08:
Should I use -c: v cedrus264 and the default is h264?
In motion too it is necessary source codes to correct prompt please

JamBalaya56562, 2026-11-09:
As ccrisan said, Motion encodes through the ffmpeg libraries, so a hardware encoder has to be in those libraries. Current versions offer H.264/OMX and H.264/V4L2M2M under Movie Format when ffmpeg supports them; other encoders like cedrus264 are not listed. Closing.

closed 2026-11-09 by JamBalaya56562