motioneye-project/motioneye #401: MEye saves images locally but will not upload to either Dropbox or Google Drive

Fader187, 2016-12-22:
Hi

I am having issues getting the MEye OS to upload its saved / recorded images to both Dropbox and Google Drive. I have tried both as I thought it was something I was doing wrong initially with Drive but it seems like it may be something else at fault.

I have also had a look through the log files but cant seem to see anything suspicious? Anyone have any idea where I may be going wrong?

Thanks in advance

Chris

ccrisan, 2016-12-22:
Please attach your motioneye.log file.

Fader187, 2016-12-22:
Hi Calin

Many thanks. I am very new to this whole idea so am a bit of a beginner. Please see attached log file as requested.

Regards

Chris

[motioneye (2).zip](https://github.com/ccrisan/motioneye/files/669297/motioneye.2.zip)

ccrisan, 2016-12-22:
Your log file shows no errors related to uploading. Are you sure you have configured the upload service correctly? Can you post a screenshot of your storage settings?

Fader187, 2016-12-29:
Hi Calin

Merry Xmas and a happy new year to you! Thanks for taking the time to look at this for me. Please see attached snapshot of the config. I entered the Auth key for both Google Drive and Dropbox, tested both and they both say successful but nothing uploads.

Regards

Chris
[image]

ccrisan, 2016-12-30:
Can I also see how the Still Images and Movies sections look like?

Fader187, 2016-12-30:
Sure. Please see attached. Something else i noticed is that I am not getting any email notifications for Motion detection either... does this help possibly? 
[image]

ccrisan, 2016-12-30:
Everything seems to be looking fine. And are you sure you actually have media files generated at least locally? If yes, could you please turn on debugging in Expert Settings, let it generate some media files and attach the new motioneye.log file.

Fader187, 2016-12-31:
Hi Calin

Yes all motion detection photos are getting saved locally as I can see them all via FTP connection. 
[motioneye.zip](https://github.com/ccrisan/motioneye/files/679607/motioneye.zip)

Still no email notifications or uploads appearing on Dropbox.

Regards

Chris

ccrisan, 2016-12-31:
In that case, could you please also attach your motion.log file? There might be some relaying issues there.

Fader187, 2016-12-31:
Hi again. Ok Calin, I have attached this log file in here for you.

[motion.zip](https://github.com/ccrisan/motioneye/files/679798/motion.zip)

Regards

Chris

ccrisan, 2016-12-31:
Your system hasn't generated any media files since you last applied any settings (that's the lifetime of your motion.log file which doesn't contain any media file events).

Fader187, 2016-12-31:
Is that maybe because I had to reboot the system before i was able to download the motion.log file? I had to reboot it as it was not allowing me to download the file from the GUI? If so, shall I just go trigger the motion detection again and resend the log file?

Regards

Chris

ccrisan, 2016-12-31:
Indeed, rebooting will recreate a new motion.log file.

Fader187, 2016-12-31:
Ok. I will get the new log file and post up here tomorrow Calin. In the mean time, please enjoy your new years celebrations.

Regards

Chris

Fader187, 2017-01-02:
Hi Calin

I have included here the motioneye and motion log files. Something else I noticed is that I often have to reboot the Pi to be able to download the motion.log file from the GUI.

Regards

Chris
[Logs.zip](https://github.com/ccrisan/motioneye/files/681004/Logs.zip)

ccrisan, 2017-01-02:
The reason why you can't download your motion.log file is that you're currently with debug enabled, and the log file gets huge. My advice is to turn the debugging off when not needed, as it will fill up your SD card.

I'm sorry but I can't figure out why it doesn't work. It should but it doesn't. At this point, my only idea is to try to configure everything from scratch and see if the problem persists.

P.S. I removed your log files attachment because they contained some sensitive data.

JamBalaya56562, 2026-11-09:
The logs never showed why, and there was no reply after the advice to set it up from scratch in 2017. The upload code and the Google Drive and Dropbox sign-in have changed since. Closing; please open a new issue if uploads fail on a current version.

closed 2026-11-09 by JamBalaya56562