motioneye-project/motioneye #402: Add a new RTSP camera by GUI failed but by hand-edit work

dattanchu, 2016-12-24:
If I add a new Network camera of the form rtsp://user:password@192.168.0.10:88/videoMain for my Foscam 9821W, the GUI will not recognize that it is supported. This means that I can't click on the Ok button to finish setting up.

If I already have the camera set up in MJPEG, I can then change the /etc/motioneye/thread-X.conf to change the URL from my MJPEG url to the above URL. This will work fine and allows me to up my framerate to approximately 8FPS (max set at 10) on all 10 of my cameras.

This looks to be a bug in the UI.

ccrisan, 2016-12-24:
1. What does the add dialog say when you try to add the camera? A screenshot would help.
2. What version of motion do you use (`motion -h`)?

dattanchu, 2016-12-24:
There is no error in the dialog, here is what the dialog would look like. Please ignore the ... in some of the boxes. It's from a Chrome plugin.

[image]

I am running motion 4.0.1 Copyright 2000-2016 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers

ccrisan, 2016-12-24:
Ah, username and password values have dedicated fields; you must use those instead of embedding them into the URL.

closed 2016-12-24 by ccrisan

dattanchu, 2016-12-25:
Awesome. Thanks. Changing username and password to the correct field work.