motioneye-project/motioneye #403: saved video - render issues | won't open

ericjanvanputten, 2016-12-28:
hi, motioneyeOS rules!
I'm encountering accross several camera's to 1 server location via motion eye that some files get corrupted somehow. attached 2 files. 

pi 3 with 20161125 version. 
maybe this is related to 'disconnecting' camera's. (other issue)
[Security_recordings.zip](https://github.com/ccrisan/motioneye/files/675741/Security_recordings.zip)

ericjanvanputten, 2016-12-28:
I turned off 2 of 3 camera's and the 1 camera started to record correct videos. 

Is it safe to assume that this might be a performance issue of the Pi 3?

ccrisan, 2016-12-29:
I don't think so. Nevertheless, you should open this issue with [the motion project](https://github.com/Motion-Project/motion) and see if the know about corrupted videos. You should also share a motion debug log with them as well as your motion/thread config files.

closed 2016-12-29 by ccrisan

ericjanvanputten, 2017-01-10:
the corrupt videos seem to have stopped mostly.