motioneye-project/motioneye #404: Problem to install Motioneye on Ubuntu 16.04LTS

richardsith, 2016-12-30:
I've a problem to install Motioneye on Ubuntu 16.04Lts Server, that one is a VM on VMware Fusion.
I saw the guide here
[here](https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu)

but after the step 3 when I run the pip command receive that error:

**$: pip install motioneye**
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
thanks a lot

richardsith, 2017-01-07:
any suggest??

chrisf4lc0n, 2017-01-07:
[http://stackoverflow.com/questions/36394101/pip-install-locale-error-unsupported-locale-setting](http://stackoverflow.com/questions/36394101/pip-install-locale-error-unsupported-locale-setting)

richardsith, 2017-01-15:
the error has been resolved running these commands
```
**$: export LC_ALL=C**
**$: pip install motioneye-0.35.1.tar.gz**
Processing ./motioneye-0.35.1.tar.gz
Collecting tornado>=3.1 (from motioneye===0.35.1)
  Downloading tornado-4.4.2.tar.gz (460kB)
    100% |################################| 
......
......
......
```
then I've continued with the steps suggested on site 

[here](https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu)

close the issue

richardsith, 2017-01-15:
another errors have been resolved in this way 

step5:
```
**$: sudo mkdir -p /etc/motioneye
$: sudo cp .local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf**
```

Step7:
Ubuntu 16.04Lts 
```
**$: sudo cp /home/<user>/.local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
$: sudo systemctl daemon-reload
$: sudo systemctl enable motioneye
$: sudo systemctl start motioneye**
```

now I can close that issue as resolved.

closed 2017-01-15 by richardsith

chrisf4lc0n, 2017-01-15:
All good then!

stressdidri, 2017-08-01:
Executing /lib/systemd/systemd-sysv-install enable motioneye
insserv: warning: script 'K01motioneye' missing LSB tags and overrides
insserv: warning: script 'K01vncserver' missing LSB tags and overrides
insserv: warning: script 'motioneye' missing LSB tags and overrides
insserv: warning: script 'vncserver' missing LSB tags and overrides
update-rc.d: error: motioneye Default-Start contains no runlevels, aborting.
root@ThinkCentre-M55:~#