motioneye-project/motioneye #405: MMAL Support in motioneye

kutysam, 2017-01-01:
Hi!

Since mmal support is already built into motion v4 onwards,
it would be nice if we are able to select an option to enable mmal instead of v4l2

+ mmalcam_name vc.ril.camera
- videodevice /dev/video0
in the autogenerated thread-1.conf

kutysam, 2017-01-02:
I was about to do some tests to check on certain things such as memory usage etc.

Anyways, putting this aside,
is it possible for users to 'remove' commands generated by motioneye ?
Under Extra Motion Options

I do understand that Extra Motion Options are to add commands and not to remove existing ones generated by MotionEye

ccrisan, 2017-01-02:
You can't "remove" options as they are used internally by motionEye. Without them, some of the functionalities displayed in the settings panel wouldn't be applied anymore. What do you want to achieve?

kutysam, 2017-01-22:
I think we can close this. Basically I wanted to write mmalcam_name vc.ril.camera and remove the v4l2 but unfortunately, I can't. I have to do it manually.

jasaw, 2017-09-01:
Don't close this issue just yet. I'm experimenting with MMAL to see what performance gain we can get out of it. From motion documentation, it appears that we can get the GPU to do image rotation and flipping so there's no need to do it in CPU.
```
; mmalcam_name vc.ril.camera
; mmalcam_control_params -hf
```

ccrisan, 2017-09-01:
Guys, if the only reason why we want MMAL support is the RPi-specific camera control, we need `v4l2-ctl` for that purpose. This is what `v4l2-ctl --list-ctrls` says on a RPi machine with attached CSI camera module:

User Controls

                         brightness (int)    : min=0 max=100 step=1 default=50 value=50 flags=slider
                           contrast (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                         saturation (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                        red_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                       blue_balance (int)    : min=1 max=7999 step=1 default=1000 value=1000 flags=slider
                    horizontal_flip (bool)   : default=0 value=0
                      vertical_flip (bool)   : default=0 value=0
               power_line_frequency (menu)   : min=0 max=3 default=1 value=1
                          sharpness (int)    : min=-100 max=100 step=1 default=0 value=0 flags=slider
                      color_effects (menu)   : min=0 max=15 default=0 value=0
                             rotate (int)    : min=0 max=360 step=90 default=0 value=0
                 color_effects_cbcr (int)    : min=0 max=65535 step=1 default=32896 value=32896

    Codec Controls

                 video_bitrate_mode (menu)   : min=0 max=1 default=0 value=0 flags=update
                      video_bitrate (int)    : min=25000 max=25000000 step=25000 default=10000000 value=10000000
             repeat_sequence_header (bool)   : default=0 value=0
                h264_i_frame_period (int)    : min=0 max=2147483647 step=1 default=60 value=60
                         h264_level (menu)   : min=0 max=11 default=11 value=11
                       h264_profile (menu)   : min=0 max=4 default=4 value=4

    Camera Controls

                      auto_exposure (menu)   : min=0 max=3 default=0 value=0
             exposure_time_absolute (int)    : min=1 max=10000 step=1 default=1000 value=1000
         exposure_dynamic_framerate (bool)   : default=0 value=0
                 auto_exposure_bias (intmenu): min=0 max=24 default=12 value=12
          white_balance_auto_preset (menu)   : min=0 max=9 default=1 value=1
                image_stabilization (bool)   : default=0 value=0
                    iso_sensitivity (intmenu): min=0 max=4 default=0 value=0
               iso_sensitivity_auto (menu)   : min=0 max=1 default=1 value=1
             exposure_metering_mode (menu)   : min=0 max=2 default=0 value=0
                         scene_mode (menu)   : min=0 max=13 default=0 value=0

    JPEG Compression Controls

                compression_quality (int)    : min=1 max=100 step=1 default=30 value=30

JamBalaya56562, 2026-10-28:
MMAL cameras have been supported since #631 by @inodlite (merged in October 2017): motionEye detects them with `vcgencmd` and offers them as "Local MMAL camera", writing `mmalcam_name` to the motion config. Note that MMAL only works with the legacy camera stack, which has been removed from Raspberry Pi OS since Bookworm; current camera modules are used through libcamera, see #2683 and #2812. Closing as completed.

closed 2026-10-28 by JamBalaya56562