motioneye-project/motioneye #406: Cant see csi local camera

Ikitissimo, 2017-01-01:
Hi,
i installed motion eye on a raspbian raspberry 3 model b following your installation guide without much trouble but i do not see the csi camera module attached to the pi.
*i added the bcm2835_v4l2 module to the etc/modules file
*gpu memory is default at 64
*raspbian is upgraded and updated
*firmware is at the newest version
*the camera module is the V2 without IR Filter

what am i missing or should i check?
thank you in advance

ccrisan, 2017-01-01:
Does `raspistill -o cam.jpg` work?

Ikitissimo, 2017-01-01:
dont tell anyone but i simply forgot to enable the csi camera on the raspi config..

now i have found and added the camera but unfortunately i got the message "unable to open video device". I tried to modify the camera settings in the video device advanced configurations but still nothing. What can i check?

thank you again

ccrisan, 2017-01-01:
Well this... indicates a lower level problem in communicating with the camera. Make sure you have at least  128 MB of ram for the GPU and that the camera actually works (is not faulty). Otherwise, please seek help on the Raspberry PI forums.

JamBalaya56562, 2026-10-30:
Closing this old one: it is about the legacy camera stack (`bcm2835_v4l2`, `raspistill`), which current Raspberry Pi OS no longer has, and the hardware question was answered at the time. For a CSI camera on a current Pi, see #2683 and #2812 (libcamerify). Please open a new issue if you still have trouble with a current setup.

closed 2026-10-30 by JamBalaya56562