motioneye-project/motioneye #407: How to handle brightness

veames, 2017-01-02:
Hi, What is the current recommended method to manage varying light conditions. In the past I have used auto-brightness with no issues, but it seems now I just get the wavering brightness effect.

I am using motioneye (latest) under raspbian (latest) on a Pi Model A.

I would use motionos, but I require GPIO libraries as I am gathering temp and other data to be displayed on the camera images.
TIA

jat80, 2017-01-11:
I also have the problem of the wavering brightness effect.  Is this a bug?

monkeyboyohyeah, 2017-03-13:
Same here. 

I get an image that is completely washed out - I set the brightness to ~25% again - it jolts back to perfect - then 1 day later it's back to washed out. Very annoying.

veames, 2017-03-31:
Currently the issue seems worse with snow conditions. even changing brightness and contrast doesn't seem to help with the whiteout effect in the snowy areas.

prutsky, 2017-04-06:
I noticed that up to a resolution of 800x600 the brightness is okay. Higher resolution result in to bright images on clear sunny days.

veames, 2017-04-06:
@prutsky It's funny, I just discovered the same thing two days ago, brightness is completely normalized in 800x600 and below. @ccrisan is this some hardware limitation?

ccrisan, 2017-04-07:
@veames, it could be related to video mode vs stills mode. See [this](https://github.com/raspberrypi/linux/issues/1754#issuecomment-265990072) comment on how to tweak the threshold between the two modes.

veames, 2017-04-07:
Wow! Initial tests show this fix is working! Thanks @ccrisan!
` sudo chmod 777 /var/lib/motion/`

closed 2018-05-17 by vvallee