motioneye-project/motioneye #408: ftp screenshot and video to nas

bigpaolo, 2017-01-02:
Hy ccrisan,
you have done a wonderful program.
I use it on a Raspberry pi 3 as an NVR to command 4 ip camera (3 normal ip camera and 1 onvif ip camera).

I should be grateful il you can improve your code with the possibility to save the screenshots and video via ftp to your own nas into your lan. 
Why?  
Beacause upload screenshot and video of 4 or more cameras into icloud (google drive, dropbox or something else...) needs a large band of the adsl. You must have a fibra adsl (10mega up and 100 mega down), because with a normal adsl (0,5mega up and 10mega down) is to much and you can't do anithing else on the web...

Thanks for all

Happy new year

Paolo

aoyue2702, 2017-01-02:
good idea

Sent from the interwebs

chrisf4lc0n, 2017-01-03:
You just need to map drives locally, then you just point to the folder you have mapped the remote drive/folder.

bigpaolo, 2017-01-03:
Thanks chris.
Can you make us an example please?
On my cpu i have hdd on drive C and D,
Dvdrom on drive E anf F.
My nas is on my lan.

Thanks

Paolo

chrisf4lc0n, 2017-01-03:
https://wiki.samba.org/index.php/Mounting_samba_shares_from_a_unix_client

bigpaolo, 2017-01-05:
Thanks
It works

Paolo

chrisf4lc0n, 2017-01-08:
Can we close this issue, then?

donnlee, 2017-01-09:
If your NAS supports NFS (most do), linux on the Pi works great with NFS shares.
My motioneyeOS cam mounts a NFS share when it boots and writes all pics & video to it.
That NAS also syncs with Dropbox, so I have local and remote pics & videos completely automated.

bigpaolo, 2017-01-09:
Thanks to all.
Can close the post.
Please insert the ftp save into the program or show this explanation in the wiki.

Paolo

bigpaolo, 2017-02-23:
Hy,
after a Raspberry reboot,, motioneye don't recognise the nas and start save on the internal sd,
so i have the sd full...:

pi@raspberrypi3:~ $ df -h
File system     Dim. Usati Dispon. Uso% Montato su
/dev/root              14G   14G          0   100% /
devtmpfs            459M     0      459M      0% /dev
tmpfs                  463M   16K    463M      1% /dev/shm
tmpfs                  463M   47M   416M    11% /run
tmpfs                   5,0M  4,0K      5,0M      1% /run/lock
tmpfs                  463M     0      463M      0% /sys/fs/cgroup
/dev/mmcblk0p6   66M   21M     46M    31% /boot
tmpfs                    93M     0         93M      0% /run/user/1000
/dev/mmcblk0p5   30M  443K     28M      2% /media/pi/SETTINGS

how can i do to continue saving on my nas also after a reboot?

Thanks 

Paolo

ccrisan, 2017-02-23:
Is this motionEye or motionEyeOS?

chrisf4lc0n, 2017-02-23:
Can you also post your /etc/fstab content if that is the MotionEye?

bigpaolo, 2017-02-26:
Hy,
I use MOTIONEYE (not OS)

Thin night i post my fstab

Paolo

ccrisan, 2017-02-26:
@bigpaolo this is not related to motionEye and there's motionEye can or should do about it.

closed 2017-02-26 by ccrisan