motioneye-project/motioneye #409: Problem with rtsp camera

GitHubUser1210, 2017-01-02:
I have a FOSCAM FI9900P that use only rtsp protocol, unfortunately, after adding the camera, the stream is a grey screen with the message "UNABLE TO OPEN VIDEO DEVICE".

I checked the log file and it seems that motioneye doesn't pass the right credentials to ffmpeg. I recorded successfully a stream video directly with ffmpeg (credentials are username:admin and no password). The /var/log/motion.log:

[0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.2:88/videoMain)
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 401 Unauthorized
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.2:88/videoMain): Server returned 401 Unauthorized (authorization failed)
[1:ml1] [WRN] [ALL] motion_init: Could not fetch initial image from camera Motion continues using width and height from config file(s)
[1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.2:88/videoMain)
[1:ml1] [ERR] [ENC] ffmpeg_avcodec_log: method DESCRIBE failed: 401 Unauthorized
[1:ml1] [ERR] [NET] netcam_rtsp_open_context: unable to open input(rtsp://192.168.1.2:88/videoMain): Server returned 401 Unauthorized (authorization failed)

How I can fix it? Thanks for you help.

ccrisan, 2017-01-03:
I can see the error *401 Unauthorized* so I assume there must be a problem with your credentials. If you're sure you have used the correct username and password, please open this issue in [the motion project](https://github.com/Motion-Project/motion).

msiegy, 2017-10-31:
I have the same problem, but with a different camera when using authenticated RTSP streams. I am able to view the stream in VLC as well as when using tinycam for android, So I believe i am using the correct credentials and URL. Happy to help debug.

Stream URL used: rtsp://192.168.1.123/StreamingSetting?version=1.0&action=getRTSPStream&ChannelID=1&ChannelName=Channel1

Also tried with :554 and with inline credentials:
rtsp://192.168.1.123:554/StreamingSetting?version=1.0&action=login&userName=admin&password=admin123&getRTSPStream&ChannelID=1&ChannelName=Channel1

ccrisan, 2017-11-01:
@msiegy please [report a motion issue](https://github.com/ccrisan/motioneye/wiki/Report-Motion-Issue).

closed 2017-11-01 by ccrisan