motioneye-project/motioneye #410: Can't find RPi v2 camera?

czarina, 2017-01-03:
Hi,

I am unable to get motioneye to recognize my RPi v2 camera on my raspbian RPi Zero. As you can see here, there is no camera to select.

[image]

- I followed the instructions here https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
- I ran 'sudo modprobe bcm2835-v4l2' 
- My camera is enabled and I successfully tested it using raspivid/raspistill

ccrisan, 2017-01-03:
What does `motion -h` say?

czarina, 2017-01-03:
Sorry, looks like I didn't install Motion properly. Thanks :)

czarina, 2017-01-03:
Another question - how do I configure settings like fps/resolution/rotation for the motioneye stream? I wasn't able to find documentation of this in the wiki.

ccrisan, 2017-01-03:
Have you enabled advanced settings?

czarina, 2017-01-03:
I had not. This gives me some of what I need, although there are some more advanced changes I'd like to make (video stabilization would be great, if available) - is there a full list of available options somewhere?

ccrisan, 2017-01-03:
That kind of options is not available with motion. For that, you'd either need to use motionEyeOS in Fast Network Camera mode (and lose motion detection, notifications & so on) or try other Raspberry PI camera-specific solutions.

closed 2019-06-21 by ccrisan