motioneye-project/motioneye #411: Image motion detection works.....Video trigger not

njordan77, 2017-01-03:
Hi, I have a new installation with the tiny little RPI A+ and the PiCam2 + another network camera via http-mjpg....both work and also the image trigger seems to work as files are stored into the DropBox Cloud....for both cameras....however as I try to enable/on the Video triggered by Motion Detection....nothing works.....

Unsure if I missed something...i at least tried/believe that all required codecs have been installed during setup phase...maybe I missed something....please help.

Futher question....what happens to the local storage location If I add a dropbox location to store files.....will local one be ignored for the future, I would hope that as my local installation does not have too much spacce to rotate files all the time...so I prefer that if dopbox is setup it is the only place where files are stored....is that thinking correct?

[image]
[motioneye.zip](https://github.com/ccrisan/motioneye/files/683429/motioneye.zip)

chrisf4lc0n, 2017-01-04:
Adding your device as a simple MJPEG camera instead of as a network camera will improve the framerate, but no motion detection, picture capturing or movie recording will be available for it.

njordan77, 2017-01-04:
thanks for the feedback, but simple camera does not help as my major usecase is motion detection and storing those videos related to detection. Again, my issue is that still images work either dropbox or local, but no videos are stored after motion detection is triggered off......side information....work schedule is OFF, hope OFF means that it works all the time. Regards Norbert

njordan77, 2017-01-04:
....one further observation, changing the video from motion triggered to continuous recording...still no files in the dropbox/locally stored. So its not a problem of motion detection triggering. Regards

njordan77, 2017-01-05:
some further observation.....if I do a continuous video, still there is no output.....so trigger is not the issue here....i changed from mp4 to avi...and now continuous video works....did not yet try triggered.....is there anything necessary to support the more compressed formats like mp4?
d

chrisf4lc0n, 2017-01-05:
Yup, you need to have the right libraries installed. FFMPEG should do it,
have you installed it properly for your system? From experience I know it
is awkward to install it properly with everything working on certain Linux
Distros.

njordan77, 2017-01-05:
hm, that somehow was also my idea....however i at least believe(d) that i had it installed...initially i tried by standard apt-get install ffmpeg....this provided following error log: E: Package 'ffmpeg' has no installation candidate.....

of course i did google and found out that it has been replaced/integrated in a bigger package....have to check which one it was....but i did then install this and was sure it would be all right....

Can you give me a hint what is necessary for Raspberry PI (raspbian jessie)....as ffmpeg is not existing anymore in standard package list coming out of the box with raspbian. Regards Norbert

chrisf4lc0n, 2017-01-05:
Have you tried Calin's pre-compiled package?
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

njordan77, 2017-01-06:
Thanks....i tried the package, according to the notes also removed som previously installed packages....but in the end nothing worked anymore...not showing videos anymore in the GUI. So i did a 180degree turn and tried out the OS-version of motioneye...which in generally i do not like (to much predefined)...but i gave it a try and I have to say that even if it cuts of some functionality/freedom of Linux but it was quite handsome......now also mp4 works....BUT still motion detection is not working at all...also with the internal camera of PI (v2) it simply ignores any frame change....i set up a 3% change and disabled the light change to 0%....so it should immediately record if I cover 30% of the picture...but nothing happens.....so no i really start to get frustrated...any idea what still could be wrong? Thanks a lot.

chrisf4lc0n, 2017-01-06:
Is there a chance you could do a fresh Raspbian installation?
I know it could be a pain, but if it works you may want to give it a shot.

njordan77, 2017-01-06:
OK, but not sure if you got my message above....i completely reset'ed yesterday and took motioneyeOS instead....whatelse can be done that is more back to the roots and taking what the dev-team provides without any adaptation.....and this is what frustrates a little...as everything is factory now and still the PICAM and the external network camera do not react to motion detections......at all.

Does the schedule have to be set maybe, as its OFF and i hope this means that it works all the time.....elsewise i have no furhter idea...but again frustrating as motioneye looks very nice but if the main feature of motion detection does not work it will not help me...sadly.

Regards
Norbert

JamBalaya56562, 2026-11-07:
Closing, as this is from 2017: mp4 movies were missing because the ffmpeg build lacked the codec, while avi worked, and the later motionEyeOS attempt was not followed up. Today `motioneye_init` installs a Motion package from the Motion releases or the distribution, as the README describes. If movies or motion detection still fail with a current release, please open a new issue with the motionEye and Motion logs.

closed 2026-11-07 by JamBalaya56562