motioneye-project/motioneye #414: Archer T2UH usb

HolyDriver, 2017-01-06:
Hello

Is there any way to use tp-link AC600 with motioneye?
The old tl-wn722n works fine with extra "wpa_supplicant.conf" file.

Thank you

chrisf4lc0n, 2017-01-07:
MotionEye or MotionEyeOS?

HolyDriver, 2017-01-08:
MotionEyeOS

What is the difference bettween them?

chrisf4lc0n, 2017-01-08:
MotionEyeOS is an operating system.
MotionEye is a software which you install on the Linux Distro of your choice.

HolyDriver, 2017-01-09:
Ok. tp-link AC600 with motioneyeOS?

chrisf4lc0n, 2017-01-09:
@HolyDriver I think what he meant by that is to connect the actual machine running MotionEye/MotionEyeOS to that router via wifi.

HolyDriver, 2017-01-10:
The raspberry pi(with motioneyeOS) can't recognize the wifi usb dongle(tp-link AC600) because there are not drivers for it. Can I install the drivers someway?

chrisf4lc0n, 2017-01-10:
If you are skillful enough yes, but you would need to build it into the MotionEyeOS... With MotionEye ran on a a different Linux Distribution that could be much easier.

HolyDriver, 2017-01-10:
Could someone build it into the MotionEyeOS? Otherwise, is there any tutorial to do it myself, at a different Linux Distribution?

JamBalaya56562, 2026-11-07:
Closing, as this asks for a Wi-Fi driver in motionEyeOS, which is built in its own repository (motioneye-project/motioneyeos), not here. As noted above, motionEye installed on a regular Linux distribution can use whatever drivers that distribution provides.

closed 2026-11-07 by JamBalaya56562