motioneye-project/motioneye #417: new Camera ID field

carnighi, 2017-01-11:
Hi,
I have added my new ipcam on motioeye sucessfully, so it recognizes the cam, and I can see the video stream, but when i try to change the cam configuration through the settings:

[image]

The "Camera ID" field remain empty and it is read only!
If i try to apply the new settings, motioneye show me this message:

[image]

Someone can help me? 

I've already tried to modify manually the conf file to add the missing field value without success. :(

Thank you!

ccrisan, 2017-01-11:
Please clear your browser's cache.

carnighi, 2017-01-11:
It works! :) Thank you very much!

closed 2017-01-11 by carnighi