motioneye-project/motioneye #418: Text Overlay size and date format

jat80, 2017-01-11:
hi there,

Is there anyway to make the text overlay font larger?

Also where does it get the date format from?  It's currently YYYY/MM/DD.  I want it to be DD/MM/YYYY.

Thanks,

JT.

chrisf4lc0n, 2017-01-11:
You will need to edit the [https://github.com/ccrisan/motioneye/blob/master/motioneye/config.py](https://github.com/ccrisan/motioneye/blob/master/motioneye/config.py)
Change:
`elif text_left == '%Y-%m-%d\\n%T':`
to:
`elif text_left == '%d-%m-%Y\\n%T':`
for the left side of the text, for the right one the relevan entry with _elif text_right_....
To have it in the DD/MM/YYYY format and change:
`'text_double': False,`
to:
`'text_double': True,`

jat80, 2017-01-11:
Thanks chrisf4lc0n.  Is it possible to change it while motionEye is already installed on my raspberry pi3?

chrisf4lc0n, 2017-01-12:
Yup, just edit
`/usr/local/lib/python2.7/dist-packages/motioneye/config.py`
locally.
The only problem is that when you upgrade to the newest version from the git master it will get overwritten to wha it was before.

jat80, 2017-01-13:
Perfect! Thanks chrisf4lc0n.

closed 2017-01-13 by jat80