motioneye-project/motioneye #419: upgrading motioneye gives following error

ArmandH, 2017-01-11:
Hi,

Trying to upgrade i see the following error, don't know if its known or relevant?

Installing collected packages: jinja2, pillow, olefile
  Found existing installation: Jinja2 2.7.3
    Not uninstalling Jinja2 at /usr/lib/python2.7/dist-packages, owned by OS
Compiling /tmp/pip-build-GBXFBA/jinja2/jinja2/asyncfilters.py ...
  File "/tmp/pip-build-GBXFBA/jinja2/jinja2/asyncfilters.py", line 7
    async def auto_to_seq(value):
            ^
SyntaxError: invalid syntax

Compiling /tmp/pip-build-GBXFBA/jinja2/jinja2/asyncsupport.py ...
  File "/tmp/pip-build-GBXFBA/jinja2/jinja2/asyncsupport.py", line 22
    async def concat_async(async_gen):
            ^
SyntaxError: invalid syntax

  Found existing installation: Pillow 2.6.1
...

Successfully installed jinja2 pillow olefile
Cleaning up...

Seems to end of succesfully

chrisf4lc0n, 2017-01-14:
Does it render anything unusable?

ArmandH, 2017-01-14:
Till now everything seems to work

afedulov, 2017-05-14:
Seems like the version of Jinja that is being pulled uses asyncio functionality, which is only available since python 3.6

JamBalaya56562, 2026-11-09:
Those were only byte-compile warnings for Jinja2's async modules under Python 2; the upgrade itself succeeded and worked. motionEye now runs on Python 3.9 or newer, where this does not happen. Closing.

closed 2026-11-09 by JamBalaya56562