motioneye-project/motioneye #420: Hikvision RTSP strem does not work

sirLeone, 2017-01-11:
Hi
I have very popular ip cam hikvision DS-2CD2335-I and i am able to use its RTSP stream in different application but cant on motioneye. I have a fresh installation of motioneye according to Debian installation guide, cam can be added to motioneye without errors with stream :

rtsp://192.168.2.XX:554/Streaming/Channels/1

[image]

Unfortunately there is no image - only grey screen : 
[image]

Logfile tells that url is invalid which is not true :) because i am using exactly same url for other application. 
My cam config for main stream is :
1920x1080P
20fps
H.264

```
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.2.XX:554/Streaming/Channels/1)
[1] [CRT] [NET] netcam_start: Invalid netcam_url (rtsp://192.168.2.XX:554/Streaming/Channels/1)
```

Do you have any ideas what could be wrong ?

ccrisan, 2017-01-11:
Please post this issue to the [motion](https://github.com/Motion-Project/motion) issue tracker.

closed 2017-01-11 by ccrisan

MarcWoodyard, 2020-12-29:
Hi, I hate to reopen closed threads but I encountered the same issue today and found a solution. 

Newer Hikvision cameras by default use digest authentication for RTSP streams. You'll need to change it to use basic authentication by logging into the camera and going to Configuration -->System --> Security. There will be a dropdown field labeled "RTSP Authentication", click on the drop-down and select the option "digest/basic" and it should work.

However, if you tried the solutions above and it didn't work, you could disable authentication for the RTSP stream (option only available on older Hikvision cameras) (**Security Warning:** Only do this if your camera feed is on a local network, **DO NOT do this if your camera is accessible over the internet because anyone will be able to view the video stream.)**

cicos99, 2022-12-29:
Had same issue .. but.. I faund an half solution ..
First I setup a camera with stream ch1, not work, delete, new camera with stream 2, not work , delete.
After I created 2 Camera , N°1 with stream channel 1 , N°2 with stream channel 2 , one of two is working ,  disabled not working !

soulraven, 2023-01-16:
how this ticket is fixed? i have the same problem with DS-7208HUHI-K1

MiG-41, 2023-04-15:
Same with HIKVISION DS-2CD2T46G2-2I .... Changing to digest/basic not helped...

dirkmueber, 2024-10-17:
Hiki2035
for 640x480 20 fps h264 512Kb:
rtsp://192.168.178.21/Streaming/Channels/2
for 1920x1080 25 fps h264 2Mb 
rtsp://192.168.178.21/Streaming/Channels/1
edit resolution manual in settings