motioneye-project/motioneye #421: Q: multiple motions + common storage + one gui?

ikke-t, 2017-01-11:
Hi,

is there any ways/plans to have multiple motions to store videos into NAS, and one GUI to let control it all?

The difference being that the current documentation describes a way that video is all the time streamed over network, and one motion tracks the movement from those several remote cameras. I'd rather not stream all the time, but let raspi boxes notice and store the motion into NAS, and only host the gui in e.g. NAS box (FreeNAS in my case). Kind of like the original motionbrowser does, but with several sources.

The motioneye looks really nice interface, I'd like to use it, but not with one motion engine, from NAS + several motion raspis instead.

Any recommendations?

ikke-t, 2017-01-11:
This is the MotionBrowser which works without streaming: http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionBrowser

ccrisan, 2017-01-12:
I'm not sure what you are trying to achieve but here's what I can tell you about motionEye(OS):
 * you can tell motionEye to upload movies to a common SMB NAS, you don't need to actually see the stream in the UI all the time
 * you can add as many cameras as your hardware allows on one single instance of motionEye
 * there's [this](https://github.com/ccrisan/motioneyeos/wiki/Usage-Scenarios) article that may help you set up multiple cameras of various types, all talking to one motionEye instance

ikke-t, 2017-01-12:
Thanks for answer @ccrisan. Still to verify:

> you can tell motionEye to upload movies to a common SMB NAS, you don't need to actually see the stream in the UI all the time

So can I set one motionEye software on the central server, with no local cameras, and point it to local SMB storage, and it would list the movies?

I already have two Motion enabled boxes (odroid+raspi) pushing motion movies to SMB, can I somehow put motionEye in front of them to my NAS server so that it would utilize the recordings coming from the existing boxes?

ikke-t, 2017-01-13:
Sorry, only now I read through the documentation fully, and found the correct term I'm looking for: Media Browser.

So I installed motionEye on my NAS box. I connected the both motion devices I have, as cameras which stream live video fine to motionEye GUI. So as the motion detected videos are already stored on the NAS over NFS, can I somehow use the motionEye to use them as media library via media browser for the configured camera?

ccrisan, 2017-01-13:
Of course you can use the media browser. Just make sure you add your cameras as Network Cameras (not Remote motionEye camera, nor Simple MJPEG Camera) and configure the following on your motionEye:
 * the motion detection mechanism for each of them
 * the File Storage is set to your NAS for all of your cameras

Then just use the motionEye media browser for the selected camera.

ikke-t, 2017-01-14:
I installed now ME on NAS server, freebsd jail. Two RasPis are running ME locally, with cameras. Both RasPis have network storage configured from NAS over cifs. They store the videos fine.

It seems the main ME sitting on the freebsd has no way knowing the videos are already on it. Instead the videos traverse my wlan twice while being downloaded. When I download a video:

1. raspi starts to pull it from NAS via cifs
2. my laptop starts to pull it from RasPi via http

While the RasPi 1 is dead slow, this video takes rather long to pull. Is there any way to let the motionEye sitting in NAS to know that video is already there, dont' route it through RasPi?

Naturally, the NAS could see the videos, if there is just an option to tell it that the videos are collected centrally, and are local. Like adding an option to tell the path to locally stored videos taken from remote cameras.

Any option to have such feature?

ccrisan, 2017-01-15:
Not really. My solution to this problem is to attach the HDD locally to the central motionEye and getting rid of the NAS. Otherwise, your movies will have to travel twice. I see no other way.

ikke-t, 2017-01-15:
That's what I'm looking for. This NAS is the central server, it's capable of running ME central server. So how do I point the ME to local copies of the movies on central server?

ccrisan, 2017-01-15:
Well, you add all of your cameras as network cameras to the central motionEye and instruct each one of them to use the local storage (in the File Storage section).

ikke-t, 2017-01-15:
That would cause continuous streaming in wlan, and all motion load to move from raspis to central server, right?

ccrisan, 2017-01-15:
Yeap, that's the obvious downside, but not if you're filming everything, like I do, not just motion-detected movies.

Otherwise you could create a NAS out of your central motionEye, configure your individual motionEye cams to use it as NAS and create one or more fake cameras on your central motionEye just to be able to browse those files "locally".

ikke-t, 2017-01-15:
That fake part is the thing I'm after here. I do record only motions. The best solution for such would be that while adding new remote ME camera, it would have a checkbox asking for local central storage path for the recirdings of remote camera. I suppose that would do it? So not fake camera, but an additional option for the ME remote camera setup.

Would you be willing to add such? I suck at web languages, but I'd be glad to test it and do what I can to help.

I suppose it would require changes:
1. option for central storage path at ccentral server in remote camera setup
2. change in media buttons while going after remote camera media. If central storage set, stream from local storage (on central server) like it's done in all in one setup, instead of remote stream.

The code for point two is already existing, as it's basically what is done normally without central server. It would just need some decision logic to be done based on the "central server media path" option.

How do you feel about it?

ccrisan, 2017-01-15:
What you are saying makes sense but to be honest I don't have the necessary time and resources to take care of such a feature these days. I'll leave this issue open anyway :)

heggink, 2018-01-14:
Hey ikke-t, you mentioned you installed motioneye under a freebsd, right (ie not in a linux VM)? Can you tell me how you did that? I would love to run my camera's as simple network cameras and have motioneye on freenas do the motion processing.

ikke-t, 2018-01-14:
Hi @heggink, I have had central MotionEye GUI on FreeNAS for a while. I mainly use it directly from RasPi and O-Droid which have cameras. It should work the same on the latest one too. My jail is rather old.

Anyhow, spin up a new jail, nowadays you would do it via iocage, or the new FreeNAS GUI. Mine is still from the times of warden. Then go in and verify with pkg install that you have the required dependencies from here: https://github.com/ccrisan/motioneye/wiki/Install-On-Other-Distributions

The follow the instructions with Pip. I believe that was it. I don't exactly remember, I did it one evening long ago. I don't recall it requiring any tricks, should be easy. This is with no remote cameras connected to this motioneye directly, as I don't wan't to fill my wifi with constant stream of video. Instead I use separate devices (Raspi with motioneye, and ODroid with plain motion) to do the motion detection locally, and store the captured video on SMB share on FreeNAS over WLAN. It works, but it is slow to use the central motioneye GUI on FreeNAS, due all video packets traverse the WLAN so many times while using the central GUI.

I never fixed the init script for it, I just launch it with:
  meyectl startserver -b -c /usr/local/etc/motioneye/motioneye.conf 

With this "storage aware" feature MotionEye would be excellent tool on FreeNAS. Or any NAS for that matter. Probably works also OK as such while using copper network instead of WLAN.

heggink, 2018-01-14:
Since I am not an expert on freebsd, took me a bit to get everything installed but it works brilliant. Thank!!!

developerfromjokela, 2019-05-19:
> Hi @heggink, I have had central MotionEye GUI on FreeNAS for a while. However I mainly use motioneye directly from RasPi and O-Droid which have local cameras. It should work the same on the latest one too. My jail is rather old.
> 
> Anyhow, spin up a new jail, nowadays you would do it via iocage, or the new FreeNAS GUI. Mine is still from the times of warden. Then go in and verify with pkg install that you have the required dependencies from here: https://github.com/ccrisan/motioneye/wiki/Install-On-Other-Distributions
> 
> The follow the instructions with Pip. I believe that was it. I don't exactly remember, I did it one evening long ago. I don't recall it requiring any tricks, should be easy. This is with no remote cameras connected to this motioneye directly, as I don't wan't to fill my wifi with constant stream of video. Instead I use separate devices (Raspi with motioneye, and ODroid with plain motion) to do the motion detection locally, and store the captured video on SMB share on FreeNAS over WLAN. It works, but it is slow to use the central motioneye GUI on FreeNAS, due all video packets traverse the WLAN so many times while using the central GUI.
> 
> I never fixed the init script for it, I just launch it with:
> meyectl startserver -b -c /usr/local/etc/motioneye/motioneye.conf
> 
> With this "storage aware" feature MotionEye would be excellent tool on FreeNAS. Or any NAS for that matter. Probably works also OK as such while using copper network instead of WLAN.

> Since I am not an expert on freebsd, took me a bit to get everything installed but it works brilliant. Thanks!!!

Hello!

I also tried to install it, but failed. How did you installed it in freenas jail?

ikke-t, 2019-05-19:
See prevoius comment from me: https://github.com/ccrisan/motioneye/issues/421#issuecomment-357539662

developerfromjokela, 2019-05-19:
@ikke-t Yes, I checked it out, but I am running into issues, 1. Dependencies are not found via pkg install (I ran pkg update before installing),
2. Can't install pip. Running Freenas 11.2-U4.1 with Intel Xeons E5440 and 24 GB of RAM.

ikke-t, 2019-05-19:
Unfortunately I don't have any new tips for you. I haven't run it since on FreeNAS, as it doesn't make sense to me. If this ticket on day gets picked up, it would make sense for me too. And I can't update to 11.2 FreeNAS, as it won't boot on my old HW due some freebsd bootloader bug.

Perhaps you need to look for the dependencies under a bit different name in BSD repos? Linux packaging might have a bit different naming compared to BSD, it's likely that. Try searching similar names for packages, or use freebsd ports web gui to find them. I'm sorry but I don't have interest to try installing it ATM.

ascl00, 2020-01-19:
I realise this is a bit old, but the package you want is py27-pip:
# pkg install py27-pip

Note however that sometimes jails get created with crippled repository, you may need to check /etc/pkg/FreeBSD.conf is pointing to the right place.

MichaIng, 2026-11-01:
#3508 adds proper FreeBSD support.

closed 2026-11-01 by MichaIng