motioneye-project/motioneye #422: endless login when adding a camera

jhmenke, 2017-01-14:
Hello,
yesterday i installed the current version of motioneye on my _14.04. Ubuntu server_. After logging in as admin i wanted to add a camera. I put in the required data and clicked Ok.
_Next the empty login screen appears_. Now regardless of what i enter in that login screen it says _"Invalid Credentials"_. I have tried the admin and user account. Also i tried the login data for the camera (which was found to be correct in the configuration form of the camera).

It seems my admin login is not persistent. Also the camera was not saved. I have tried Google Chrome, Microsoft Edge, Firefox and Vivaldi to make sure it was not a cookie related or Adblock problem. The behaviour is consistent.

What can i do to fix this?

chrisf4lc0n, 2017-01-14:
Logs please...

jhmenke, 2017-01-14:
where do i find them? /etc/motioneye and /var/lib/motioneye do not have any log files.

chrisf4lc0n, 2017-01-14:
/var/log/motion.log
for starters

jhmenke, 2017-01-14:
here is the relevant part. it seems the authentication failed despite me being logged in before.

2017-01-14 10:56:39: [motioneye]  WARNING: 403 GET /config/list/?_=1484387797831&scheme=http&host=xxx&port=xxx&path=%2F&username=xxx&password=xxxproto=netcam&_username=admin&_login=true&_signature=c2d3327dc5e67f9497bacbf9bf66d5eda5c60f69 (93.208.180.131) 0.68ms
2017-01-14 10:56:40: [motioneye]    ERROR: authentication failed for user admin

chrisf4lc0n, 2017-01-14:
Can you reset the camera?

jhmenke, 2017-01-14:
Yes. I have cut power and restarted it, no difference. But the problem seems to be the motioneye server with the 403 error?!

chrisf4lc0n, 2017-01-14:
Restart may not be enough, try factory reset...
Also try to upgrade motion to 4.0.1.

chrisf4lc0n, 2017-01-14:
If it is the RTSP camera new Motion may cure it... I know that the MotionEye/Motion are not running on the camera, hence I was asking to factory reset the camera and try default login/password credentials.

closed 2018-06-04 by jhmenke