motioneye-project/motioneye #423: Whitebalance and ISO shutter-time

dfect, 2017-01-14:
Hi,

I use motioneye with the raspi-cam (the new NOIR-8MP version) and a raspberry 3. Often the streamed and recorded video and photos become very blue in tone and very dark. Is there some way to modify the White Balance or shutter time or some other way improving the dark areas present?

Using the following command takes natural color and more brighten areas but I havnt figured out if its possible with motioneye.

sudo raspistill -o picture.jpg --timeout 1000 --shutter 100000 --ISO 100000 -awb off -awbg 0.88,0.97

chrisf4lc0n, 2017-01-14:
You can play around with brightness and contrast by adding it to 'Extra Motion Options'
Like:
`brightness 124
contrast 34`

[http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions)

dfect, 2017-01-14:
Thanks, I've tried different brightness, contrast, hue and v4l2_palette settings, all seems to affect but in other ways. Seems hard to adjust the actual shutter time in each photo.

chrisf4lc0n, 2017-01-14:
You can in fact use imagemagick to post process the photo.
This is what I use to flip the photo:
`convert -rotate 180 /home/pi/433Utils/RPi_utils/$t$d.jpg /home/pi/433Utils/RPi_utils/$t$d.jpg`
but imagemagick gives you much more:
[https://www.imagemagick.org/script/command-line-processing.php](https://www.imagemagick.org/script/command-line-processing.php)

JamBalaya56562, 2026-11-09:
This was answered here (Extra Motion Options, or post-processing with ImageMagick), and it was about the legacy raspistill camera stack, which current Raspberry Pi OS no longer has. Closing; please open a new issue if you need this with a current camera setup.

closed 2026-11-09 by JamBalaya56562