motioneye-project/motioneye #425: no motioneye front-end on Ubuntu 16.04Lts Server Edition

richardsith, 2017-01-15:
After I've followed the guide

[here](https://github.com/ccrisan/motioneye/wiki/Installation)

and correct some errors with this post

[here](https://github.com/ccrisan/motioneye/issues/404)

the motioneye has been correctly installed, but I've a problem using 

http://[your_ip]:8765/

I don't see anything. Any suggest??

chrisf4lc0n, 2017-01-15:
What port have you set up in motioneye.conf and is your IP set to 0.0.0.0?
`sudo nano /etc/motioneye/motioneye.conf`

richardsith, 2017-01-15:
@chrisf4lc0n the port is 8765

chrisf4lc0n, 2017-01-17:
What is the actual error when you are trying to load the page up?

richardsith, 2017-01-19:
@chrisf4lc0n the result is that

[image]

chrisf4lc0n, 2017-01-19:
Can you ping that address?

richardsith, 2017-01-19:
yes

chrisf4lc0n, 2017-01-19:
When you do:
`netstat --listen`
what output do you get for port 8765?

richardsith, 2017-01-19:
that the result:

```
$: netstat --listen
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 *:ssh                   *:*                     LISTEN     
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN     
udp        0      0 *:bootpc                *:*                                
Active UNIX domain sockets (only servers)
Proto RefCnt Flags       Type       State         I-Node   Path
unix  2      [ ACC ]     STREAM     LISTENING     19776    /run/user/1000/systemd/private
unix  2      [ ACC ]     SEQPACKET  LISTENING     11600    /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     11596    /run/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     14910    /var/lib/lxd/unix.socket
unix  2      [ ACC ]     STREAM     LISTENING     11601    /run/lvm/lvmetad.socket
unix  2      [ ACC ]     STREAM     LISTENING     11602    /run/lvm/lvmpolld.socket
unix  2      [ ACC ]     STREAM     LISTENING     11614    /run/systemd/fsck.progress
unix  2      [ ACC ]     STREAM     LISTENING     11615    /run/systemd/journal/stdout
unix  2      [ ACC ]     STREAM     LISTENING     14940    /run/snapd.socket
unix  2      [ ACC ]     STREAM     LISTENING     14941    /run/snapd-snap.socket
unix  2      [ ACC ]     STREAM     LISTENING     14942    /run/acpid.socket
unix  2      [ ACC ]     STREAM     LISTENING     14943    /run/uuidd/request
unix  2      [ ACC ]     STREAM     LISTENING     14948    /var/run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     16962    @ISCSIADM_ABSTRACT_NAMESPACE
```

any idea?

chrisf4lc0n, 2017-01-19:
It should something along the lines of:
tcp        0      0 *:8765                  *:*                     LISTEN

Do:
ps ax | grep motioneye
ps ax | grep motion

and post the results...
I suspect the motioneye is not running at all.

richardsith, 2017-01-19:
I've run this
$: sudo systemctl start motioneye

and the result is that:

:ps ax | grep motioneye
  1508 pts/0    S+     0:00 grep --color=auto motioneye
--------------------------------------
Ubuntu 16.04Lts Video Hub Server
:ps ax | grep motion
  1510 pts/0    S+     0:00 grep --color=auto motion

that service are not running, why?

chrisf4lc0n, 2017-01-19:
`sudo nano /etc/motioneye/motioneye.conf`
change log to debug and post motioneye.log content.

richardsith, 2017-01-20:
report its content:

```
$: more /etc/motioneye/motioneye.conf 

# the base URL path at which motionEye lives
# change this if you run motionEye behind a reverse proxy (e.g. nginx),
# and you want to make motionEye accessible at a specific base path (e.g. /cams)
#base_path /cams

# path to the configuration directory (must be writable by motionEye)
conf_path /etc/motioneye

# path to the directory where pid files go (must be writable by motionEye)
run_path /var/run

# path to the directory where log files go (must be writable by motionEye)
log_path /var/log

# default output path for media files (must be writable by motionEye)
media_path /var/lib/motioneye

# path to the motion binary to use (automatically detected if commented)
#motion_binary /usr/bin/motion

# the log level (use quiet, error, warning, info or debug)
log_level info

# the IP address to listen on
# (0.0.0.0 for all interfaces, 127.0.0.1 for localhost)
listen 0.0.0.0

# the TCP port to listen on
port 8765

# whether motion HTTP control interface listens on
# localhost or on all interfaces
motion_control_localhost true

# the TCP port that motion HTTP control interface listens on
motion_control_port 7999

# interval in seconds at which motionEye checks the SMB mounts
mount_check_interval 300

# interval in seconds at which motionEye checks if motion is running
motion_check_interval 10

# whether to restart the motion daemon when an error occurs while communicating with it
motion_restart_on_errors false

# interval in seconds at which the janitor is called
# to remove old pictures and movies
cleanup_interval 43200

# timeout in seconds to wait for response from a remote motionEye server
remote_request_timeout 10

# timeout in seconds to wait for mjpg data from the motion daemon
mjpg_client_timeout 10

# timeout in seconds after which an idle mjpg client is removed
# (set to 0 to disable)
mjpg_client_idle_timeout 10

# enable SMB shares (requires motionEye to run as root) 
smb_shares false

# the directory where the SMB mount points will be created
smb_mount_root /media
[... 405 lines cut ...]
------------------------
cap.driver: "uvcvideo"
cap.card: "VMware Virtual USB Video Device"
cap.bus_info: "usb-0000:02:03.0-1"
cap.capabilities=0x84200001
------------------------
[1:ml1] [NTC] [VID] [Jan 15 19:59:07] v4l2_get_capability: - VIDEO_CAPTURE
[1:ml1] [NTC] [VID] [Jan 15 19:59:07] v4l2_get_capability: - STREAMING
[1:ml1] [NTC] [VID] [Jan 15 19:59:07] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00000000
[1:ml1] [NTC] [VID] [Jan 15 19:59:07] v4l2_select_input: - CAMERA
[1:ml1] [ERR] [VID] [Jan 15 19:59:07] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
[1:ml1] [NTC] [VID] [Jan 15 19:59:07] vid_v4lx_start: Using V4L1
[1:ml1] [NTC] [ALL] [Jan 15 19:59:07] image_ring_resize: Resizing pre_capture buffer to 1 items
[1:ml1] [NTC] [EVT] [Jan 15 19:59:11] event_new_video Source FPS 2
[1:ml1] [NTC] [ENC] [Jan 15 19:59:11] ffmpeg_open Selected Output FPS 2
[1:ml1] [NTC] [EVT] [Jan 15 19:59:11] event_newfile: File of type 8 saved to: /var/lib/motion/01-20170115195911.avi
[1:ml1] [NTC] [ALL] [Jan 15 19:59:11] motion_detected: Motion detected - starting event 1
[1:ml1] [NTC] [EVT] [Jan 15 19:59:11] event_newfile: File of type 1 saved to: /var/lib/motion/01-20170115195911-01.jpg
[1:ml1] [NTC] [EVT] [Jan 15 19:59:12] event_newfile: File of type 1 saved to: /var/lib/motion/01-20170115195912-00.jpg
[1:ml1] [NTC] [EVT] [Jan 15 19:59:12] event_newfile: File of type 1 saved to: /var/lib/motion/01-20170115195912-01.jpg
[1:ml1] [NTC] [EVT] [Jan 15 19:59:13] event_newfile: File of type 1 saved to: /var/lib/motion/01-20170115195913-00.jpg
[1:ml1] [NTC] [EVT] [Jan 15 19:59:13] event_newfile: File of type 1 saved to: /var/lib/motion/01-20170115195913-01.jpg
```

dd

chrisf4lc0n, 2017-01-20:
Change log_level to debug and post motioneye.log content please...

richardsith, 2017-01-20:
in motion.conf I've changed the level

# Level of log messages [1..9] (EMG, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). (default: 6 / NTC)
log_level 9

where is the motioneye.log? I've only /var/log/motion/motion.log......

chrisf4lc0n, 2017-01-20:
If it is not in /var/log/ as per your config file it is not running. Are you sure the startup script is correct and it does what it should?
sudo systemctl status motioneye

richardsith, 2017-01-20:
```
sudo systemctl status motioneye
[sudo] password for richardsith: 
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-01-20 10:25:14 CET; 21min ago
  Process: 887 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=
 Main PID: 887 (code=exited, status=203/EXEC)

Jan 20 10:25:14 videohub systemd[1]: Started motionEye Server.
Jan 20 10:25:14 videohub systemd[1]: motioneye.service: Main process exited, code=exited, status=203/EXEC
Jan 20 10:25:14 videohub systemd[1]: motioneye.service: Unit entered failed state.
Jan 20 10:25:14 videohub systemd[1]: motioneye.service: Failed with result 'exit-code'.
```

chrisf4lc0n, 2017-01-20:
Error 203 normally means there are errors in the startup script.

chrisf4lc0n, 2017-01-20:
Refer to wiki and try to run motioneye manually with the *.conf file you using.

richardsith, 2017-01-20:
```
$: meyectl startserver -c /etc/motioneye/motioneye.conf 
    INFO: hello! this is motionEye server 0.35.1
CRITICAL: config directory "/etc/motioneye" does not exist or is not writable
--------------------------------------
Ubuntu 16.04Lts Video Hub Server
:ls -la /etc/motion*
/etc/motion:
total 56
drwxr-xr-x  2 root root  4096 Jan 20 10:24 .
drwxr-xr-x 96 root root  4096 Jan 20 09:22 ..
-rw-r--r--  1 root root  2753 Oct 25 04:15 camera1-dist.conf
-rw-r--r--  1 root root  2752 Oct 25 04:15 camera2-dist.conf
-rw-r--r--  1 root root  2753 Oct 25 04:15 camera3-dist.conf
-rw-r--r--  1 root root  2753 Oct 25 04:15 camera4-dist.conf
-rw-r--r--  1 root root 29028 Jan 20 10:24 motion.conf

/etc/motioneye:
total 12
drwxr-xr-x  2 root root 4096 Jan 20 10:17 .
drwxr-xr-x 96 root root 4096 Jan 20 09:22 ..
-rw-r--r--  1 root root 2858 Jan 15 20:00 motioneye.conf
```

chrisf4lc0n, 2017-01-20:
chmod 7777 motioneye.conf

richardsith, 2017-01-20:
the same prb

```
Ubuntu 16.04Lts Video Hub Server
:sudo chmod 7777 /etc/motioneye/motioneye.conf 
[sudo] password for richardsith: 
--------------------------------------
Ubuntu 16.04Lts Video Hub Server
:meyectl startserver -c /etc/motioneye/motioneye.conf 
    INFO: hello! this is motionEye server 0.35.1
CRITICAL: config directory "/etc/motioneye" does not exist or is not writable
--------------------------------------
Ubuntu 16.04Lts Video Hub Server
:sudo chmod -R 7777 /etc/motioneye
--------------------------------------
Ubuntu 16.04Lts Video Hub Server
:meyectl startserver -c /etc/motioneye/motioneye.conf 
    INFO: hello! this is motionEye server 0.35.1
CRITICAL: pid directory "/var/run" does not exist or is not writable
```

richardsith, 2017-01-21:
re-making all steps and the result was the same, the commands I used are:

```
$: sudo apt update
$: apt list --upgradable
$: sudo apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev
$: wget https://pypi.python.org/packages/02/c2/cfdadf641821f511a0fb29ab81762732ec7c84f822ff766677508af6682d/motioneye-0.35.1.tar.gz#md5=0a0182773fb178a8f7a95bf5b34b4e0b
$: sudo export LC_ALL=C
$: sudo pip install motioneye-0.35.1.tar.gz
$: sudo mkdir -p /etc/motioneye
$: sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.con
$: sudo mkdir -p /var/lib/motioneye
$: sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
$: sudo systemctl daemon-reload 
$: sudo systemctl enable motioneye 
$: sudo systemctl start motioneye 
$: sudo meyectl startserver -c /etc/motioneye/motioneye.con
    INFO: hello! this is motionEye server 0.35.1
    INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
    INFO: interrupt signal received, shutting down...
    INFO: server stopped
    INFO: tasks stopped
    INFO: bye!
```

but now

ccrisan, 2017-01-22:
It's motioneye.conf, not motioneye.con.

closed 2017-01-22 by ccrisan

chrisf4lc0n, 2017-01-22:
Ubuntu 16.04Lts Video Hub Server
:meyectl startserver -c /etc/motioneye/motioneye.conf 
    INFO: hello! this is motionEye server 0.35.1
CRITICAL: pid directory "/var/run" does not exist or is not

s0rent, 2017-01-22:
The issue is very real and not invalid.
I tried yesterday to setup a 16.04 server and got the same error when following the guide. No access to /var/run and /var/log and the motioneye server wont start. Setting them to chmod 7777 can make it start but it is not a viable solution.

chrisf4lc0n, 2017-01-22:
Change location of the .log .conf and .pid to your home directory and
report back.

s0rent, 2017-01-22:
That makes the manual start possible (meyectl startserver) but the startup script still fails

chrisf4lc0n, 2017-01-22:
There must be some problem with permissions along the way... Are you
running MotionEye as root?

s0rent, 2017-01-22:
Since i didnt edit the systemd script i guess it runs as root by default? I just followed the guide and i dont know how motioneye works internally

richardsith, 2017-01-22:
this is the systemd script:

```
$: more /etc/systemd/system/motioneye.service 
[Unit]
Description=motionEye Server

[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort

[Install]
WantedBy=multi-user.target
```
as I already posted, using sudo the result is that

```
$: sudo meyectl startserver -c /etc/motioneye/motioneye.conf 
sudo: meyectl: command not found
```

s

chrisf4lc0n, 2017-01-22:
In Service section add
User=root
and run it again.

richardsith, 2017-01-22:
```
[Unit]
Description=motionEye Server
[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort
User=root
[Install]
WantedBy=multi-user.target
```
run it again
```

$: meyectl startserver -c /etc/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
    CRITICAL: config directory "/etc/motioneye" does not exist or is not writable
```

or with root

```
$: sudo meyectl startserver -c /etc/motioneye/motioneye.conf
sudo: meyectl: command not found
```

chrisf4lc0n, 2017-01-22:
What about if you run the service script you have just edited?

richardsith, 2017-01-22:
`sudo chmod -R 777 /etc/motioneye/`
then
```
meyectl startserver -c /etc/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
CRITICAL: pid directory "/var/run" does not exist or is not writable
```

chrisf4lc0n, 2017-01-22:
Since you have edited the systemd startup script, why don't you run it?

`sudo systemctl start motioneye`

richardsith, 2017-01-22:
any errors

chrisf4lc0n, 2017-01-22:
`sudo systemctl status motioneye`

richardsith, 2017-01-22:
sudo systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2017-01-22 17:47:52 CET; 1min 10s ago
  Process: 1504 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exite
 Main PID: 1504 (code=exited, status=203/EXEC)

Jan 22 17:47:52 motion systemd[1]: Started motionEye Server.
Jan 22 17:47:52 motion systemd[1]: motioneye.service: Main process exited, code=exited, status=203/EXEC
Jan 22 17:47:52 motion systemd[1]: motioneye.service: Unit entered failed state.
Jan 22 17:47:52 motion systemd[1]: motioneye.service: Failed with result 'exit-code'.
lines 1-10/10 (END)

chrisf4lc0n, 2017-01-22:
I give up!

richardsith, 2017-01-22:
noooooo!!!!!

s0rent, 2017-01-22:
the start up script fails regardless of whether you explicitly set the user to root or any other user. I suppose its easier if you start a virtual machine with ubuntu server 16 and test it if you have the time.

richardsith, 2017-01-22:
never give up :-) ......I've done another test, using  the guide [here](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User) as normal user and I received the same issue

```
$: meyectl startserver -c /home/richardsith/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
    INFO: main config file /home/richardsith/motioneye/motion.conf does not exist, using default values
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
```

????

richardsith, 2017-01-22:
I tried to
`$:  mv /etc/motion/motion.conf  /home/richardsith/motioneye/ 
`then run the script obtained
```
$: meyectl startserver -c /home/richardsith/motioneye/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
```
on my browser the page is finally that

[image]

s0rent, 2017-01-22:
The issue is not to start it, the issue is that the automatic startup script isnt working.

richardsith, 2017-01-22:
the script is that:

```
[Unit]
Description=motionEye Server
[Service]
ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Restart=on-abort
[Install]
WantedBy=multi-user.target
```
but seeing this screen the meyectl is not there

[image]

richardsith, 2017-01-22:
I also tried this
[image]

`$: mv /etc/motioneye/motioneye.conf /etc/motion/`

then run the script obtained

```
$: meyectl startserver -c /etc/motion/motioneye.conf
    INFO: hello! this is motionEye server 0.35.1
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
$: systemctl status motioneye
 motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2017-01-22 20:28:47 CET; 5s ago
 Main PID: 1513 (meyectl)
    Tasks: 5
   Memory: 71.6M
      CPU: 277ms
   CGroup: /system.slice/motioneye.service
           ├─1513 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motion/motioneye.conf
           └─1524 /usr/bin/python /usr/local/bin/meyectl startserver -c /etc/motion/motioneye.conf

Jan 22 20:28:47 motion systemd[1]: Stopped motionEye Server.
Jan 22 20:28:47 motion systemd[1]: Started motionEye Server.
Jan 22 20:28:47 motion meyectl[1513]:     INFO: hello! this is motionEye server 0.35.1
Jan 22 20:28:47 motion meyectl[1513]:     INFO: cleanup started
Jan 22 20:28:47 motion meyectl[1513]:     INFO: wsswitch started
Jan 22 20:28:47 motion meyectl[1513]:     INFO: tasks started
Jan 22 20:28:47 motion meyectl[1513]:     INFO: mjpg client garbage collector started
Jan 22 20:28:47 motion meyectl[1513]:     INFO: server started

```
on my browser the page is finally that

s0rent, 2017-01-23:
Can we at least agree that this issue is not irrelevant?

richardsith, 2017-01-23:
if you run the pip command to install motioneye with local user the installer creates that path (/home/username/.local/bin), while if you use sudo  the path is /usr/local/bin/

bomellberg, 2017-12-07:
For future users:

Please make sure that, if you use the ubuntu firewall (ufw), allow port 8765:

ufw allow 8765

Without this, I was not able to reach the web front end.