motioneye-project/motioneye #427: FTP server

oaf2015, 2017-01-19:
Is there any way to utilize motioneye with FTP servers (Filezilla or something as such)? Are Google drive and Dropbox the only 2 options? I appreciate your help.

oaf2015, 2017-01-19:
Do you have any instructions for the central server usage scenario?

chrisf4lc0n, 2017-01-19:
The answer to your 1st question:
https://github.com/ccrisan/motioneye/issues/392

What do you mean by central server usage scenario?

oaf2015, 2017-01-19:
I am trying to upload to my own FTP server rather than Google drive or Dropbox but do not know how to go about it under the GUI.

chrisf4lc0n, 2017-01-19:
No option in GUI, you have to map SFTP drive locally and point the MotionEye to save files to it.

oaf2015, 2017-01-19:
Under the .conf file?

oaf2015, 2017-01-19:
Do you have an example of the command?

chrisf4lc0n, 2017-01-19:
What OS?

oaf2015, 2017-01-19:
Jessie

chrisf4lc0n, 2017-01-19:
[https://www.digitalocean.com/community/tutorials/how-to-use-sshfs-to-mount-remote-file-systems-over-ssh](https://www.digitalocean.com/community/tutorials/how-to-use-sshfs-to-mount-remote-file-systems-over-ssh)

oaf2015, 2017-01-19:
Great I will try it out. Thanks!

closed 2019-06-21 by ccrisan