motioneye-project/motioneye #428: Motion triggered movies.

billbeau, 2017-01-20:
The motion triggered movies look like a still image. The files are mp4 and VLC opens them but you can tell by looking at the clock in the movie that the time does not change. The seconds are frozen. These files are in the megabits so they are movies.

billbeau, 2017-01-20:
Never mind. I found it. It was in VLC I had to change one of the preferences. Never had to do that for motion before. Sorry Folks.

mikecarr, 2017-01-25:
what was the setting you had to change?

billbeau, 2017-01-26:
I had to change the preferences for video to automatic

JamBalaya56562, 2026-11-07:
Closing, as you found the cause: the movies looked frozen because of a video preference in VLC, and setting it to automatic fixed it.

closed 2026-11-07 by JamBalaya56562